ARMmbed/mbed-os #4586: TARGET_STM does not have ethernet abstraction layer like TARGET_NXP has

peeyushpashine, 2017-06-20:
In target.json
[image]
For few targets they have both LWIP and ethernet(mbed) abstraction. While for STM it is not yet present. Are there future plans for the same?

bcostm, 2017-06-22:
Hi,

I think only the LWIP feature is used to enable the Ethernet. @0xc0170 do you confirm ?

I checked the targets.json and the ETHERNET (in "device_has" field) is no more present on STM boards. The LWIP feature is on the boards:

DISCO_F746NG
DISCO_F769NI
NUCLEO_F207ZG
NUCLEO_F429ZI
NUCLEO_F439ZI
NUCLEO_F746ZG
NUCLEO_F756ZG
NUCLEO_F767ZI

cc @adustm @LMESTM @jeromecoutant

geky, 2017-07-20:
- `"device_has": "ETHERNET"` is a deprecated tag we are moving away from.
- `"features": "LWIP"` enables the LWIP network stack, which is how we currently provide ethernet support. Currently this requires an implementation of the lwip `netif` api, these currently live [here](https://github.com/ARMmbed/mbed-os/tree/master/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch).
- We are also working on introducing `"device_has": "ETH"` for ethernet support in https://github.com/ARMmbed/mbed-os/pull/4079, but this is a work in progress.

@bulislaw, feel free to correct me if I'm wrong

theotherjimmy, 2017-07-21:
@bulislaw @geky:

Instead of deprecateing `"ETHERNET"` and adding `"ETH"`, why not just use `"ETHERNET"` without deprecating it?

peeyushpashine, 2017-11-20:
@geky @bcostm Does ETHERNET via LWIP is fully supported on ST targets now like the other peripherals? how are netif APIs called using ethernet abstraction? I was trying to find the bind but could not find that [here](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_STM/TARGET_STM32F7/TARGET_NUCLEO_F746ZG/stm32f7_eth_init.c):

bcostm, 2017-11-20:
> Does ETHERNET via LWIP is fully supported on ST targets now like the other peripherals? 

Yes

0xc0170, 2017-11-20:
@peeyushpashine Can this be closed?

peeyushpashine, 2017-11-21:
@0xc0170 Sure.

closed 2017-11-21 by peeyushpashine