ARMmbed/mbed-os #4587: DFUService is missing?

shayo, 2017-06-20:
- Type: Bug n
- Priority: Minor

---------------------------------------------------------------
## Bug

I think there is a missing include file somewhere. When trying to compile a project with an instance of DFUService, I get an error stating that DFUService is an unknown class.

**Target**
NRF52DK

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
latest on github

shayo, 2017-06-24:
Looks like DFUService still has 
#ifdef TARGET_NRF51822

What is the latest on official DFU support for NRF52?

shayo, 2017-07-12:
Is there a reason why the feature-nrf5_dfu_s13x_v2 branch hasn't been merged yet to the master branch?

seppestas, 2017-08-03:
The bootloader used for DFU still uses the legacy device manager iof. the peer manager causing disconnects when performing DFU.

I currently have a branch that allows to use the legacy device manager on the nRF52 target, but I think the proper solution would be to change the bootloader so it uses the peer manager (and maybe also to support signed deployment packages while we're at it).

jrobeson, 2017-09-20:
@seppestas : but some of us just want something that works now :).  Is the branch easy to rebase on top of the current master?

seppestas, 2017-09-20:
Yup, it should be pretty easy if you are familiar with handling minor merge conflicts.

seppestas, 2017-09-20:
Note that because the bootloader uses a different method for handling peer information than what is used by default in mbed-os, i.e the legacy device manager instead of the peer manager, the connection will drop and the bootloader will fall back to the old "increment mac address to trigger a characteristic discovery" method when starting the DFU. This means you have to connect to the "new" DFUTarg device to perform the actual DFU.

I fixed this by making the rest of mbed OS use the legacy device manager (being one of the people looking for a solution that works for now), however this still does not seem to work properly with bonded iOS devices.

I could create a PR for the feature-nrf5_dfu_s13x_v2 branch to incorporate these changes, but I think the best way forward is to update the bootloader.

cmonr, 2017-12-14:
Closing since NRF52_DK now supports all BLE examples. Feel free to open another issue comes up.

closed 2017-12-14 by cmonr