ARMmbed/mbed-os #4588: TRNG support for nRF5x devices

HITMAnsOFT, 2017-06-20:
- Type: Enhancement

I am working on a BLE-based car alarm using nRF52832, and I need secure random key generation for a custom communication protocol. The nRF51x and nRF52x devices all contain a hardware RNG core, and the HAL and drivers are available in the Nordic SDK. However, the mbed OS doesn't seem to support trng_api for Nordic targets. I tried to directly call the Nordic HAL methods, but I found out that the corresponding files are missing from the TARGET_SDK11 folder. Those files (drivers_nrf/hal/nrf_rng.h, drivers_nrf/rng/nrf_drv_rng.h, drivers_nrf/rng/nrf_drv_rng.c) do exist in the TARGET_SDK13 folder, but for some reason it is not used with the nRF52832 device.

It would be nice to support TRNG HAL for nRF51 and nRF52 devices.

pan-, 2017-06-20:
@nvlsianpu

HITMAnsOFT, 2017-06-26:
Update: I downloaded the Nordic SDK version 11 and copied the HAL and driver files mentioned above into my project source dir and was able to use the RNG module directly.

nvlsianpu, 2017-06-27:
@nvlsianpu have You a POC for look at?

HITMAnsOFT, 2017-06-27:
@nvlsianpu I have this small C++ wrapper. It may not be robust and may cause an infinite loop, but it's working for me on nRF52832. The buffer depth of the RNG seems to be 32 bytes. It's C++11/14, but you'll get the idea:
```
#include "SEGGER_RTT.h"
#include "nrf_drv_rng.h"

class Random {
public:
    Random ()
    {
        nrf_drv_rng_init(nullptr);
    }

    template <class T, size_t size> void getArray(std::array<T, size>& out)
    {
        for (T& a : out) {
            uint8_t available = 0;
            do {
                nrf_drv_rng_bytes_available(&available);
            } while (available < sizeof(T));
            SEGGER_RTT_printf(0, "Bytes available: %d\n", available);
            nrf_drv_rng_rand(&a, sizeof(T)); 
        }
    }
};

```

nvlsianpu, 2017-06-28:
Thanks for the snipped. I wanted to know whether rng driver work out-off-the-box for nRF52. So answer is yes.

HITMAnsOFT, 2017-06-28:
Yes, it works. Chip code is QFAA B0. Unfortunately I was not able to run my applicaiton on nRF51. Do you happen to know if mbed works on Rev. 2 nRF51822 chips?

nvlsianpu, 2017-06-28:
mbed-os apps are supposed to work since rev. 3 of nRF51822. I never tested it with rev. 2

HITMAnsOFT, 2017-07-04:
I've ordered a couple of what seems to be Rev 3 nRF51822 modules. Will test this code on them when I get them.

dlfryar, 2018-01-12:
Mbed OS SDK updates happening [here](https://github.com/ARMmbed/mbed-os/pull/5845)

marcuschangarm, 2018-03-08:
New TRNG driver has been merged to master: https://github.com/ARMmbed/mbed-os/pull/6116

closed 2018-05-18 by 0xc0170