ARMmbed/mbed-os #4589: IAR compilation Failure

akrnvm, 2017-06-20:
### Description
- Type: Bug 
- Related issue: Compilation
- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
IAR
 compilation error

 mbed compile -m xxxxx -t IAR

[Error] rtx_delay.c@32,2: [Pe079]: expected a type specifier
[Error] rtx_delay.c@32,2: [Pe260]: explicit type is missing ("int" assumed)
[Error] rtx_delay.c@32,2: [Pe141]: unnamed prototyped parameters not allowed when body is present
[Error] rtx_delay.c@32,2: [Pe130]: expected a "{"
[Error] rtx_delay.c@32,2: [Pe020]: identifier "a1" is undefined
[Error] rtx_delay.c@33,2: [Pe079]: expected a type specifier
[Error] rtx_delay.c@33,2: [Pe260]: explicit type is missing ("int" assumed)
[Error] rtx_delay.c@33,2: [Pe247]: function "__attribute__" has already been defined (at line 32)
[Error] rtx_delay.c@33,2: [Pe141]: unnamed prototyped parameters not allowed when body is present
[Error] rtx_delay.c@33,2: [Pe130]: expected a "{"
[Error] rtx_delay.c@33,2: [Pe020]: identifier "a1" is undefined
[Error] rtx_delay.c@33,11: [Pe020]: identifier "a2" is undefined
[Warning] rtx_delay.c@77,11: [Pe223]: function "__svcDelay" declared implicitly
[Warning] rtx_delay.c@77,11: [Pe188]: enumerated type mixed with another type
[Warning] rtx_delay.c@87,11: [Pe223]: function "__svcDelayUntil" declared implicitly
[Warning] rtx_delay.c@87,0: [Pe188]: enumerated type mixed with another type
[ERROR]

0xc0170, 2017-06-20:
@akrnvm What is your IAR version?

akrnvm, 2017-06-20:
IAR 7.20
checked with master branch and release 5.5.

akrnvm, 2017-06-20:
compile successfully in 5.4.7 release.

0xc0170, 2017-06-20:
For the release 5.5 - IAR 7.8 officially supported. Can you please update?

That would explain the errors

akrnvm, 2017-06-20:
ok thanks

closed 2017-06-20 by 0xc0170