ARMmbed/mbed-os #4591: Greentea with GCC_ARM

akrnvm, 2017-06-20:
### Description
- Type: Question
- Related issue: `Greentea`
- Priority: Blocker

---------------------------------------------------------------
## Bug
test case Failing:
features-tests-filesystem-mbr_block_device
tests-events-queue
tests-events-timing

**Toolchain:**
GCC_ARM

hi
i have used "default_lib": "small" in target.json and greentae is passing with mbed 5.5
but if i remove "default_lib": "small"  above  test cases are failing

my device have 16 KB RAM, is it mandatory to pass all testates in GCC arm or  is acceptable to modify the test cases (STACK size , HEAP size ,etc).

Test cases are passing in ARM and IAR tool chains.

"default_lib": "small" is acceptable in latest mebdOS 5.5 ? else whats are the alternative method

0xc0170, 2017-06-20:
If you use verbose (-V I believe) for testing, what is printed ? We would like to see what fails (what errors actually are there) ?

version of GCC_ARM?

akrnvm, 2017-06-20:
```
[1497959694.02][CONN][RXD] {{max_heap_usage;0}}
[1497959694.02][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1497959694.03][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1497959694.015000
[1497959694.03][CONN][RXD] {{end;success}}
[1497959694.03][CONN][INF] found KV pair in stream: {{end;success}}, queued...
[1497959694.05][CONN][RXD] {{__exit;0}}
[1497959694.05][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1497959694.06][HTST][INF] __notify_complete(True)
[1497959694.06][HTST][INF] __exit(0)
[1497959694.08][HTST][INF] __exit_event_queue received
[1497959694.08][HTST][INF] test suite run finished after 5.16 sec...
[1497959694.08][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1497959694.14][HTST][INF] CONN exited with code: 0
[1497959694.14][HTST][INF] No events in queue
[1497959694.14][HTST][INF] stopped consuming events
[1497959694.14][HTST][INF] host test result() call skipped, received: True
[1497959694.14][HTST][INF] calling blocking teardown()
[1497959694.14][HTST][INF] teardown() finished
[1497959694.14][HTST][INF] {{result;success}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'OK'

```

version of GCC_ARM is 5.4

0xc0170, 2017-06-20:
> version of GCC_ARM is 5.4

mbed OS 5.5 - supported GCC ARM is 6, this however might not help, let us know

pan-, 2017-06-20:
> "default_lib": "small" is acceptable in latest mebdOS 5.5 ? else whats are the alternative method

@akrnvm It is  possible to remove the overhead of `Case` objects in `tests-events-queue` by using `case_t` POD (see [this](https://github.com/pan-/mbed/blob/7a8898bb21e3770ee2b156e8a97bf5170ef4303e/TESTS/mbed_hal/ticker/main.cpp#L1985-L2080) as an example), it will save ~500 bytes. 

For other tests, I'm afraid there is not that much to do on the OS side but maybe you can reduce the footprint of your port. What target is it ?

bridadan, 2017-06-21:
The key line in that log is this one: `[1497954078.15][CONN][RXD] Operator new[] out of memory`.

The OS is reporting that you've ran out of heap space. I'd give @pan- suggestion a try.

akrnvm, 2017-06-22:
Thank you, i will check and update.

closed 2017-07-11 by akrnvm