ARMmbed/mbed-os #4600: Does Mbed support multiple image of bin files?

Archcady, 2017-06-21:
### Description
- Type:  Question
- Priority:  Major 

---------------------------------------------------------------

## Question

We are working on Realtek target REALTEK_RTL8195AM, and we want to move Realtek code from BDRAM to SDRAM so that user code space will be sufficient for mbed-client. 

If we need to move code to SDRAM, then we wanna use two linking script files (.sct / .ld / .icf) which produce two images. 

And we wonder if this is supported by mbed-os tools to have multiple link script and multiple images. 

Thanks.

0xc0170, 2017-06-21:
I don't think this is supported. How would these 2 images work?

cc @c1728p9 @theotherjimmy @sg-

theotherjimmy, 2017-06-21:
Do you really need multiple linker scripts to accomplish this? In the case of GCC_ARM, I don't think you do. When you say you want to use 2 linker scripts, do you also want 2 linker invocations?

closed 2017-06-29 by Archcady