ARMmbed/mbed-os #4602: OS2 test MBED_A21 (Call function before main) failed only with IAR

jeromecoutant, 2017-06-21:
## Bug

On master branch, test MBED_A21 is FAIL for all STM32 targets only with IAR:

MBED_A21	Call function before main (mbed_main)	NUCLEO_F091RC	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_F103RB	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_F207ZG	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_F303ZE	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_F446RE	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_F767ZI	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_L073RZ	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_L152RE	IAR	FAIL
MBED_A21	Call function before main (mbed_main)	NUCLEO_L476RG	IAR	FAIL

**Target**
STM32

**Toolchain:**
IAR

jeromecoutant, 2017-06-21:
Proposition:

diff --git a/platform/mbed_sdk_boot.c b/platform/mbed_sdk_boot.c

@@ -109,6 +109,10 @@ int __low_level_init(void)
   return 1;
 }

+void __iar_argc_argv() {
+    mbed_main();
+}
+
 #endif

0xc0170, 2017-06-21:
Thanks for reporting ! Either we can do what you proposed, but I believe better place for IAR would be in their cmain file, where we already call other init functions.

https://github.com/ARMmbed/mbed-os/blob/master/cmsis/TOOLCHAIN_IAR/cmain.S#L88 (right after argc) we can invoke mbed_main.

```
          FUNCALL __cmain, mbed_main()
        BL      mbed_main()
```

0xc0170, 2017-06-21:
I'll send a patch rightaway now, will request a review

0xc0170, 2017-06-21:
Done, pull request sent, please review.

PR #4604 (2017-06-26): cmain IAR: add mbed main
This fixes #4602 issue, mbed_main should be invoked right before the real main

Tested on nucleo l476rg, using mbed_a21 test

Sorry for the formatting, my terminal was not set properly :)
before this patch:

```
{{timeout;20}}
              {{host_test_name;default_auto}}
                                             {{description;Call function mbed_main before main}}
                {{test_id;MBED_A21}}
                                    {{start}}
                                             MBED: main() starts now!
{{failure}}
           {{end}}
```

after this patch [... cut]
commit 857ec23d (2017-06-26): cmain IAR: add mbed main

This fixes #4602 issue, mbed_main should be invoked right before the real main
commit c19defe3 (2017-10-13): cmain IAR: add mbed main

This fixes #4602 issue, mbed_main should be invoked right before the real main
commit 6eaa3840 (2017-10-20): cmain IAR: add mbed main

This fixes #4602 issue, mbed_main should be invoked right before the real main
closed 2017-06-26 by theotherjimmy