ARMmbed/mbed-os #4606: No target for STM32L496

Phara0h, 2017-06-21:
It is a pretty new MC but they have boards out for it already. I would love to see support for it and possibly the "P" version too.

0xc0170, 2017-06-22:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-06-22:
Hi
Same answer as in https://developer.mbed.org/questions/78168/Cant-compile-for-the-L496ZG-P:
There is currently no planned date to support this board at ST side...

Maybe you can check with @pmancele in #4339 
In this Issue, you can find the PeripheralPins.c file

Regards,

pmancele, 2017-06-22:
Hi,
We plan to propose a merge request adding support for the NUCLEO_L496ZG board next week based on the PeripheralPins.c provided by @jeromecoutant.

Regards

bcostm, 2017-10-16:
Can you please close this "issue" ? Thanks.

ST_TO_BE_CLOSED

PR #4650 (2017-11-02): NUCLEO_L496ZG: Add new Platform
## Description
Add support for the NUCLEO_L496ZG and the STM32L496xG MCUs.
All the toolchains (ARM_MICRO/ARM_STD/IAR/GCC_ARM) have been added but only the GCC_ARM one has been tested on our side.

This PR should resolve #4606.

## Status
**Ready**

This is our first port so feel free to point anything we should change.

Regards
closed 2017-10-18 by 0xc0170