ARMmbed/mbed-os #4609: UARTSerial baud rate change support

RobMeades, 2017-06-22:
### Description
- Type: Enhancement
- Priority: Minor

`UARTSerial()` does not currently expose the underlying `serial_baud()` functionality.  This is normally OK, however it is a problem if you want to run a modem interface at high baud rates.  A modem will auto-baud to the rate at which the MCU serial port is running but only up to a point (e.g. 115200 or sometimes 230400).  If you want to run at rates higher than that, you need to send a specific command to the modem (e.g. `AT+IPR` in the case of the u-blox modems), then switch the serial port to the new baud rate afterwards.

Without this functionality, the only way to get achieve a higher baud rate is to delete the instance of `UARTSerial()` and start it up again.  However `UARTSerial()` is used usually used by other classes, e.g. `ATCmdParser()`, and so these classes *also* have to be deleted and reinstantiated, which means losing all of their state (e.g. all of the out of band callbacks that may have been set up by the various users of that class).

So we need a `UARTSerial.set_baud()` (or thereabouts) please.

RobMeades, 2017-06-22:
Adding parties I know could be interested: @kjbracey-arm, @hasnainvirk.

deepikabhavnani, 2017-09-21:
Closing as fix is merged in PR #4615.

closed 2017-09-21 by deepikabhavnani