ARMmbed/mbed-os #4611: Realtek RTL8195A prints debug info by default for WiFi

bulislaw, 2017-06-22:
I just extended WiFi example to support RTL8195A and it works, which is great! https://github.com/ARMmbed/mbed-os-example-wifi/pull/40 

The board is printing some info on serial I'd rather not have by default thought:
```
WiFi example

Scan:
Network: Bartek secured: WPA2 BSSID: 22:6A:31:74:f0:61 RSSI: -26 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:64:3a:80 RSSI: -44 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:3a:81 RSSI: -44 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:35:21 RSSI: -48 Ch: 11
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:51:1 RSSI: -50 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:64:35:20 RSSI: -50 Ch: 11
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:64:51:0 RSSI: -52 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:8d:1 RSSI: -56 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:64:8d:0 RSSI: -56 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:64:4b:a0 RSSI: -56 Ch: 11
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:4b:a1 RSSI: -58 Ch: 11
Network: mbed secured: WPA2 BSSID: 18:D6:C7:52:5f:1e RSSI: -60 Ch: 10
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:95:c1 RSSI: -68 Ch: 1
Network: TP-LINK_4568 secured: WPA2 BSSID: 84:16:F9:94:45:68 RSSI: -68 Ch: 10
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:5e:41 RSSI: -68 Ch: 11
15 networks available.

Connecting...

RTL8195A[Driver]: set ssid [Bartek] 

RTL8195A[Driver]: start auth to 22:6A:31:74:f0:61

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=1)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
Success

MAC: f0:03:8c:3c:20:43
IP: 172.20.10.12
Netmask: 255.255.255.240
Gateway: 172.20.10.1
RSSI: -46

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

ioctl[SIOCGIWESSID] ssid = NULL, not connected
Done
```

All the lines prefixed with `RTL8195A[Driver]` and `ioctl[SIOCGIWESSID] ` should be hidden by default. I would recommend adding `debug` parameter to the constructor defaulting to false that would toggle printing of whatever extra information you see fit, but by default you shouldn't print anything (except hard errors maybe).

@Archcady @tung7970

Archcady, 2017-06-22:
RTL8195A[Driver] message is sent from inside of our lib, so we'll try to add a switch for these debug messages. Thanks for the advice.

0xc0170, 2017-06-22:
> RTL8195A[Driver] message is sent from inside of our lib, so we'll try to add a switch for these debug messages. Thanks for the advice.

They are always ON at the moment? It would be great to have it optional, and by default OFF.

Archcady, 2017-06-29:
We create this PR to mute the debug messages mentioned here. https://github.com/ARMmbed/mbed-os/pull/4667

JanneKiiskila, 2017-12-03:
I think issue can be closed with, PR #4667 is merged.

Archcady, 2017-12-15:
IS this issue still open?

JanneKiiskila, 2017-12-15:
@0xc0170  @bulislaw  - it's not open anynmore, this issue should be closed (like my comment 12 days ago already suggested).

closed 2017-12-18 by 0xc0170