ARMmbed/mbed-os #4612: Timing tests are unstable on NRF51_DK

0xc0170, 2017-06-22:
### Description
- Type: Bug 
- Priority: Blocker

# Bug

**Target**
NRF51_DK

**Toolchain:**
any

**meed-os sha:**
latest master

The test that is failing tests-mbedmicro-rtos-mbed-basic , it fails randomly, so might be harder to reproduce.

This is one https://github.com/ARMmbed/mbed-os/pull/4601#issuecomment-310222458 and another one https://github.com/ARMmbed/mbed-os/pull/4599#issuecomment-310301070

different compilers, same issue. The result from the log

```
mbedgt: test on hardware with target id: 110000005032312036393738323032313231303497969903
mbedgt: test suite 'tests-mbedmicro-rtos-mbed-basic' ................................................. TIMEOUT in 261.48 sec
	test case: 'tests-mbedmicro-rtos-mbed-basic' ................................................. ERROR in 261.48 sec
```

Test timeouts.  There are some reports for nrf51 dk and its timers, are they relevant?

cc @nvlsianpu @anangl @pan- @c1728p9

pan-, 2017-06-22:
It might be linked to #4108 but I cannot reproduce the issue locally.

0xc0170, 2017-06-22:
I got NRF51 DK working, and can reproduce the problem. Every second time I run the rtos basic test , I get this

```

[1498142884.62][HTST][INF] Device base time 72000
[1498142884.62][HTST][INF] sleeping for 9.08000946045 to measure drift accurately
[1498142884.62][SERI][TXD] {{final_time;0}}
[1498142884.76][CONN][RXD] {{final_time;CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x200055CC, task name: (null))
```

c1728p9, 2017-06-22:
I looked at a similar problem with @studavekar and we found the cause to be a stack overflow because the [768 byte stack in the basic rtos test](https://github.com/ARMmbed/mbed-os/blob/75935b466d4fe14aa42bc083b827034f932cd6e4/TESTS/mbedmicro-rtos-mbed/basic/main.cpp#L28) was too small. It intermittently failed in development builds because the stack would only overflow if an interrupt pushed more registers onto the thread stack at the right time. Rebuilding as debug caused the overflow to occur always since it used more stack space. @studavekar is updating the stack size in #4599 so this should fix the underflow in tests-mbedmicro-rtos-mbed-basic.

Also of note, the above stack overflow was not nrf51 specific, and has been occurring on other platforms.

nvlsianpu, 2017-11-27:
@xc0170,  @c1728p9  Is this issue is still valid?

0xc0170, 2017-11-27:
I believe this was resolved earlier with updating ticker tests at the host side plus above the stack overflow.

closed 2017-11-27 by 0xc0170