ARMmbed/mbed-os #4617: A number of targets are not recognised in the online IDE

adbridge, 2017-06-22:
The following targets are note recognised by the online IDE when trying to compile them.

"USENSE" is not recognized                                                                                                        
"TB_SENSE_12" is not recognized                                                                                                   "DISCO_L475VG_IOT01A" is not recognized                                                                                           "EFM32PG12_STK3402" is not recognized                                                                                             "TB_SENSE_1" is not recognized                                                                                                    
"UBRIDGE" is not recognized                                                                                                       "NRF52840_DK" is not recognized                                                                                                   "RO359B" is not recognized                                                                                                        "ARM_CM3DS_MPS2" is not recognized                                                                                                "DISCO_L072CZ_LRWAN1" is not recognized

theotherjimmy, 2017-06-22:
Were these just added? Could you give me a bit to push the tools up.

0xc0170, 2017-06-22:
Yes, most of  them are new as I can tell

adbridge, 2017-06-22:
2 or 3 have been around for a couple of months now :)

theotherjimmy, 2017-06-22:
huh, welp. Going to do a tools release anyway. We'll see if that fixes things.

deepikabhavnani, 2018-12-28:
@adbridge - Issue is still valid or can be closed?

adbridge, 2019-01-02:
@deepikabhavnani AFAIK this is still a valid issue. I will update this issue the next time I make an mbed2 release.