ARMmbed/mbed-os #4618: Single test : call before main test case is failing in LPC1343(customize target).

pranavshandilya, 2017-06-22:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug  | Question
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
LPC1347

**Toolchain:**
 IAR

**Toolchain version:**
8

**DAPLink version:**
0243

**Expected behavior**
A test case should be a pass.

**Actual behavior**
Test case is failing.

**Steps to reproduce**
1. Download mbed-os-5.5.1
2. Go to the tools directory.
3. create muts_all.json configuration and test_spec.json file content could look like below:-
```
{
    "1" : {"mcu": "LPC1343",
        "port":"COM9",
        "disk":"G:\\",
        "peripherals": [""]
    }
}
```
```
{
    "targets": {
        "LPC1343" : ["IAR"]
     }
}
```
4. Run the below command
5. python singletest.py -i test_spec.json -M muts_all.json -n MBED_A21 -v
```
Test::Output::Start
MBED: Instrumentation: "COM9" and disk: "G:\"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '20'
HOST: Property 'host_test_name' = 'default_auto'
HOST: Property 'description' = 'Call function mbed_main before main'
HOST: Property 'test_id' = 'MBED_A21'
HOST: Start test...
MBED: main() starts now!
{{failure}}
{{end}}
Test::Output::Finish
TargetTest::LPC1343::IAR::MBED_A21::Call function before main (mbed_main) [FAIL] in 1.41 of 20 sec
Test summary:
+--------+---------+-----------+----------+---------------------------------------+--------------------+---------------+-------+
| Result | Target  | Toolchain | Test ID  | Test Description                      | Elapsed Time (sec) | Timeout (sec) | Loops |
+--------+---------+-----------+----------+---------------------------------------+--------------------+---------------+-------+
| FAIL   | LPC1343 | IAR       | MBED_A21 | Call function before main (mbed_main) |        1.41        |       20      |  0/1  |
+--------+---------+-----------+----------+---------------------------------------+--------------------+---------------+-------+
Result: 1 FAIL

Completed in 9.60 sec

Build successes:
  * LPC1343::IAR::MBED
  * LPC1343::IAR::MBED_A21
```

## Question
I am unable to find the issue.
The Same test case is passing with other compilers(Like GCC_ARM, ARM).
Could you please suggest some reference?

adustm, 2017-06-22:
Hello, I have the same issue with ST platform DISCO_F746NG
`python tools/singletest.py --auto -f DISCO_F746NG --tc=GCC_ARM  -v -V -n MBED_A21` is OK
`python tools/singletest.py --auto -f DISCO_F746NG --tc=IAR  -v -V -n MBED_A21` fails
it looks like the line `HOST: Unknown property: MBED: mbed_main() call before main()` is missing in the output
So the code 

```
extern "C" void mbed_main() {
    printf("MBED: mbed_main() call before main()\r\n");
    mbed_main_called = true;
}
```
is not executed

Does anyone has an idea why ?

Kind regards

jeromecoutant, 2017-06-22:
Hi Armelle

see #4604

0xc0170, 2017-06-22:
Thanks for reporting, as noted above, this will be fixed soon on master.

pranavshandilya, 2017-06-22:
thanks, @0xc0170  :)

closed 2017-06-22 by pranavshandilya