ARMmbed/mbed-os #4619: greentea test : some test cases are failing in LPC1343

pranavshandilya, 2017-06-22:
### Description
- Type: Bug 
- Related issue: [https://github.com/ARMmbed/mbed-os/issues/4591](url)
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
LPC1343

**Toolchain:**
GCC_ARM | ARM | IAR

**Toolchain version:**
GCC_ARM : 6
ARM : 6
IAR : 8

**DAPLink version:**
0243

**Expected behavior**
All below test cases are passing in Mbed os 5.4.7
```
tests-events-queue 
tests-events-timing
tests-mbed_drivers-wait_us
tests-mbedmicro-rtos-mbed-basic
tests-mbedmicro-rtos-mbed-malloc
tests-mbedmicro-rtos-mbed-mutex 
```

**Actual behavior**
All above test cases are failing in Mbed os 5.5.1

## Question

I am unable to find the issue.
Could you please suggest me some reference?

pan-, 2017-06-22:
@pranavshandilya Can you post the test log ?

pranavshandilya, 2017-06-22:
@pan- Thanks for your quick reply :)

Below is the 2 test case logs contents for GCC ARM 

1. tests-events-timing

```
$ mbed test -m LPC1343 -t GCC_ARM -n tests-events-timing -v
Building library mbed-build (LPC1343, GCC_ARM)
Scan: mbed-os-mbed-os-5.5.1
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
[DEBUG] Macros: -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1498153138.51 -DTOOLCHAIN_object -D__CMSIS_RTOS -DTARGET_LPC1343 -DTOOLCHAIN_GCC -DTARGET_CORTEX_M -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M0 -DTARGET_NXP -DDEVICE_SERIAL=1 -D__LPC1343__ -DDEVICE_INTERRUPTIN=1 -D__CORTEX_M0 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_CORTEX_M0
Scan: GCC_ARM
Building project timing (LPC1343, GCC_ARM)
Scan: GCC_ARM
Scan: timing
[DEBUG] Macros: -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTOOLCHAIN_object -D__CMSIS_RTOS -DTARGET_LPC1343 -DTOOLCHAIN_GCC -DTARGET_CORTEX_M -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M0 -DTARGET_NXP -DDEVICE_SERIAL=1 -D__LPC1343__ -DDEVICE_INTERRUPTIN=1 -D__CORTEX_M0 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DMBED_BUILD_TIMESTAMP=1498153139.28 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_CORTEX_M0
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Fill      |    83 |    10 |   22 |
| Misc      | 74395 |  2746 | 7002 |
| Subtotals | 74478 |  2756 | 7024 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 9780 bytes
Total RAM memory (data + bss + heap + stack): 9780 bytes
Total Flash memory (text + data + misc): 77234 bytes
Image: BUILD/tests/LPC1343/GCC_ARM/TESTS/events/timing/timing.bin

Memory map breakdown for built projects (values in Bytes):
+--------+---------+-----------+------------+-------+------+-----------+-------------+
| name   | target  | toolchain | static_ram | stack | heap | total_ram | total_flash |
+--------+---------+-----------+------------+-------+------+-----------+-------------+
| timing | LPC1343 | GCC_ARM   |       9780 |     0 |    0 |      9780 |       77234 |
+--------+---------+-----------+------------+-------+------+-----------+-------------+

Build successes:
  * LPC1343::GCC_ARM::MBED-BUILD
  * LPC1343::GCC_ARM::TESTS-EVENTS-TIMING
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: test specification file 'D:\Projects\MBEDTest\MBED_5_5_1\mbed-os-mbed-os-5.5.1\BUILD\tests\LPC1343\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'D:\Projects\MBEDTest\MBED_5_5_1\mbed-os-mbed-os-5.5.1\BUILD\tests\LPC1343\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | LPC1343       | LPC1343[0]           | COM9        | F:          | 106200000000000000000000000000000000000097969902 |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: processing target 'LPC1343' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | LPC1343       | LPC1343[0]           | COM9:9600   | F:          | 106200000000000000000000000000000000000097969902 |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
[... 227 lines cut ...]
mbedgt: test suite report:
+-----------------+---------------+--------------------+--------+--------------------+-------------+
| target          | platform_name | test suite         | result | elapsed_time (sec) | copy_method |
+-----------------+---------------+--------------------+--------+--------------------+-------------+
| LPC1343-GCC_ARM | LPC1343       | tests-events-queue | FAIL   | 56.74              | shell       |
+-----------------+---------------+--------------------+--------+--------------------+-------------+
[... cut ...]
| LPC1343-GCC_ARM | LPC1343       | tests-events-queue | Testing calls with 4 args | 1      | 0      | OK     | 0.06               |
| LPC1343-GCC_ARM | LPC1343       | tests-events-queue | Testing calls with 5 args | 1      | 0      | OK     | 0.05               |
+-----------------+---------------+--------------------+---------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 FAIL / 6 OK
mbedgt: completed in 57.40 sec
mbedgt: exited with code 1
```

Kind regards

0xc0170, 2017-06-26:
cc @toyowata

toyowata, 2017-06-27:
I could not find the LPC1343 target support in the targets.json.  Is it new platform?

pranavshandilya, 2017-06-27:
@toyowata 

`I could not find the LPC1343 target support in the targets.json. Is it new platform?`
No, this is not a new platform and LPC1343 target is not present in the targets.json.

This is the custom target `http://www.zilogic.com/blog/product-zkit-arm-1343.html` which is used by me for the small applications using MbedOS.
I have ported all Hal driver (Ref. : LPC1768) and after porting all the test case is passing using MbedOS 5.4.7.
But in MbedOS 5.5.1 above test cases are failing.

Thanks

0xc0170, 2017-06-27:
I also overlooked this is not a target in our codebase.

> I have ported all Hal driver (Ref. : LPC1768) and after porting all the test case is passing using MbedOS 5.4.7.
But in MbedOS 5.5.1 above test cases are failing.

To be aligned, use IAR 7.8 version (8 not supported yet, might fail?), ARMCC 5.2x (not 6).

We have seen some failures with those, what are the failures, do you get any error from RTX (stack underflow or overflow), or just timing issues? For most of them you dont need a host test, thus create an application and you can debug it to explore the failures, or print verbose as you did with mbed test (many tests do not print that much information by default, so might tweak them a bit).

cc @c1728p9

closed 2017-07-01 by pranavshandilya