ARMmbed/mbed-os #4621: mbed-os-example-blinky breaks on NRF52_DK when NDEBUG set

saedelman, 2017-06-23:
### Description
- Type: Bug
- Related issue: None
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.2.1 20161205

**mbed-cli version:** 
1.1.1

**meed-os sha:**
1747ef4 

**DAPLink version:**

**Expected behavior**
Reduce code size

**Actual behavior**
Application hang - LED does not blink. Compile without NDEBUG set and application works as expected. 

**Steps to reproduce**
Using mBed online compiler, set NDEBUG using menu option "Compile Macros". Compile and download application to NRF52_DK. Same result with using offline compiler (set NDEBUG in mbed_app.json)

0xc0170, 2017-06-26:
cc @pan- @nvlsianpu

jrobeson, 2017-07-14:
what might be the cuase of this?

jrobeson, 2017-07-14:
oh? does the online compiler use GCC_ARM? i thought it used the arm compiler? Did that change sometime in the past year or two?

0xc0170, 2017-07-14:
> oh? does the online compiler use GCC_ARM? i thought it used the arm compiler? Did that change sometime in the past year or two?

Still ARM Compiler 

> what might be the cuase of this?

There was idle stack size set incorrectly, so might be that one https://github.com/ARMmbed/mbed-os/pull/4736

@saedelman Can you retest latest master ?

jrobeson, 2017-07-14:
@0xc0170 : I asked because the initial report said GCC_ARM. Perhaps the github issue template should have "Online Compiler" as an option for those who don't know what's running their builds.

saedelman, 2017-07-14:
@jrobeson The issue that I reported exists both with the "online compiler" as well as the "offline compiler" using GCC_ARM. 

@0xc0170 I will retest later this evening with the latest master and report back.

theotherjimmy, 2017-07-14:
@saedelman Thanks for the update. I added "Online compiler (ARMC5)" to the toolchain list.

theotherjimmy, 2017-07-14:
@saedelman You might have more luck offline by using the "release" profile. For example, you could compile with `mbed compile --profile release` That would implicitly define the `NDEBUG` macro, and turn the optimization settings of GCC to optimize for size.

saedelman, 2017-07-14:
@theotherjimmy I have done this previously and the results for the mbed-os-example-blinky are identical: it does not function on the NRF52_DK target. I will test out the latest master this evening and confirm whether this has been corrected.

pan-, 2017-07-17:
@saedelman You cannot test the tip of master in the online IDE. Even if the file `mbed-os.lib` points to the tip, the latest released version will be used instead.

I've tested it locally with the latest master and I cannot reproduce the issue. I guess it has been fixed by #4736 .

jrobeson, 2017-11-29:
This is probably fixed now isn't it? My device isn't the nrf52 dk , but is another nrf528832 device.

saedelman, 2018-01-08:
Confirmed fixed.

closed 2018-01-08 by saedelman