ARMmbed/mbed-os #4622: Nucleo-F070RB cannot build by GCC_ARM since CMSIS5 merge

mickeyln, 2017-06-23:
### Description
- Type: Bug
- Related issue: `#4459`, `#4294`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Nucleo-F070RB

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc-arm-none-eabi-5_2-2015q4-20151219-win32
gcc-arm-none-eabi-6-2017-q1-update

gcc-arm-none-eabi-5_4-2016q3 for linux

**Expected behavior**
Build finished and has bin file

**Actual behavior**
Build fail
```
../mbed-os/cmsis/core_cm0.h:591:12: error: #include expects "FILENAME" or <FILENAME>
   #include CMSIS_VECTAB_VIRTUAL_HEADER_FILE
            ^
make[1]: *** [code/ec_debug_uart.o] Error 1
make: *** [all] Error 2

```

**Steps to reproduce**
1. Export the code for GCC_ARM
2. Build by Toolchain

remove the CMSIS_VECTAB_VIRTUAL_HEADER_FILE on make and use "cmsis_nvic.h" to instate it can build pass.

theotherjimmy, 2017-06-23:
@mickeyln Thanks for reporting this. I can't seem to reproduce this problem. Could you give me some more info so that I can try to reproduce the failure? In particular I would like:
 1) What version of mbed-os you're using (I tried with 5.5.1)
 2) An example program that demonstrates this. Based on the description, it should affect anything, so blinky would be sufficient. (I tried with mbed-os-example-fat-filesystem)
 3) The C_FLAGS variable from the makefile. Mine bellow
```make
C_FLAGS += -std=gnu99
C_FLAGS += -DCMSIS_VECTAB_VIRTUAL
C_FLAGS += -D__MBED__=1
C_FLAGS += -DDEVICE_I2CSLAVE=1
C_FLAGS += -DTARGET_LIKE_MBED
C_FLAGS += -DDEVICE_PORTOUT=1
C_FLAGS += -DDEVICE_PORTINOUT=1
C_FLAGS += -D__MBED_CMSIS_RTOS_CM
C_FLAGS += -DTARGET_STM32F0
C_FLAGS += -DDEVICE_RTC=1
C_FLAGS += -DTOOLCHAIN_object
C_FLAGS += -DDEVICE_SERIAL_ASYNCH=1
C_FLAGS += -D__CMSIS_RTOS
C_FLAGS += -DTOOLCHAIN_GCC
C_FLAGS += -DDEVICE_I2C_ASYNCH=1
C_FLAGS += -DTARGET_CORTEX_M
C_FLAGS += -DTARGET_DEBUG
C_FLAGS += -DARM_MATH_CM0
C_FLAGS += -DTARGET_UVISOR_UNSUPPORTED
C_FLAGS += -DTARGET_M0
C_FLAGS += -DTARGET_STM32F070RB
C_FLAGS += -DTARGET_NUCLEO_F070RB
C_FLAGS += -DDEVICE_SERIAL=1
C_FLAGS += -DDEVICE_INTERRUPTIN=1
C_FLAGS += -D__CORTEX_M0
C_FLAGS += -DDEVICE_I2C=1
C_FLAGS += -DTRANSACTION_QUEUE_SIZE_SPI=2
C_FLAGS += -DDEVICE_STDIO_MESSAGES=1
C_FLAGS += -DTARGET_FF_MORPHO
C_FLAGS += -DDEVICE_SPI_ASYNCH=1
C_FLAGS += -DDEVICE_LOWPOWERTIMER=1
C_FLAGS += -DTARGET_FF_ARDUINO
C_FLAGS += -DTARGET_STM
C_FLAGS += -DDEVICE_SERIAL_FC=1
C_FLAGS += -DDEVICE_PORTIN=1
C_FLAGS += -DDEVICE_SLEEP=1
C_FLAGS += -DTOOLCHAIN_GCC_ARM
C_FLAGS += -DDEVICE_SPI=1
C_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"
C_FLAGS += -DDEVICE_SPISLAVE=1
C_FLAGS += -DDEVICE_ANALOGIN=1
C_FLAGS += -DDEVICE_PWMOUT=1
C_FLAGS += -DMBED_BUILD_TIMESTAMP=1498243095.89
C_FLAGS += -DTARGET_LIKE_CORTEX_M0
C_FLAGS += -include
C_FLAGS += mbed_config.h
```
 4) Was this export online or offline with mbed CLI?

mickeyln, 2017-06-26:
Hi @theotherjimmy 
Thanks for your feedback. I use the Web to export the program like the picture
[image]

and build on Ubuntu 12.04 and gcc-arm-none-eabi-5_4-2016q3 to build it.
It seems the maker can not use 
`C_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"
`

JojoS62, 2017-06-26:
C_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"
This setting causes also a problem in the exporter for SW4STM32. It adds the line
<listOptionValue builtIn="false" value="CMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h""/>
to the .cproject file and it is invalid. So the config is not read at all and eclipse complains about 'not a CDT project'. But I will open a new issue for that, not sure if this is also the reason for #3606 ?

theotherjimmy, 2017-06-26:
> It seems that the maker can not use `C_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"`

I'm sorry, I'm having trouble understanding this @mickeyln. Could you explain what you mean?

> This setting causes also a problem in the exporter for SW4STM32

Sorry about that @JojoS62 . SW4STM32 is going to be deprecated soon if it does not get supported. Please use gnuarmeclipse or eclipse_gcc_arm instead.

JojoS62, 2017-06-26:
ok, but gnuarmeclipse has the same problem, it creates the same symbol with the wrong syntax:

[image]

theotherjimmy, 2017-06-26:
Oh! dang. Let me look at that real quick.

theotherjimmy, 2017-06-26:
@JojoS62 Yep. confirmed.

theotherjimmy, 2017-06-26:
@JojoS62 I was able to get the project to recognize (by escaping), but it still does not work correctly.

JojoS62, 2017-06-26:
sorry, don't know exactly what you mean?

theotherjimmy, 2017-06-26:
I was able to export to gnuarmeclipse from a project using the nucleo-f070rd, and import the project into gnuarmeclipse. However, it still complains about no quotes in the include statement.

theotherjimmy, 2017-06-26:
Alrighty. PR in a moment.

theotherjimmy, 2017-06-26:
@mickeyln  and @JojoS62 could you try the patch in pr #4636? I think that should resolve the quoting issue you are seeing in the gnuarmeclipse exporter

JojoS62, 2017-06-26:
it worked, but there are still unecessary backslashes in the symbol value.

mickeyln, 2017-06-28:
the don't worked Makefile of export is:
```
C_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h"
CXX_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h"
ASM_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h"
```
And I change to Makefile to:
```
C_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"
CXX_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"
ASM_FLAGS += -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"
```
It is worked.

I don't know how to fix in code.

jeromecoutant, 2017-06-28:
Hi
Solution could be to remove these CMSIS_VECTAB_VIRTUAL macros in the targets.json file...

mickeyln, 2017-07-05:
Hi @jeromecoutant 
Remove these CMSIS_VECTAB_VIRTUAL macros can not finish build.
I think fix the export tool is better. It can fix all of same issues like this.
Double escape defines in gnuarmeclipse export #4636

closed 2017-07-05 by mickeyln

jeromecoutant, 2017-07-05:
> Remove these CMSIS_VECTAB_VIRTUAL macros can not finish build.

No, idea is to remove it AND update CMSIS part :-)

mickeyln, 2017-07-07:
Hi @jeromecoutant 
Yes, I agree your idea. CMSIS 5 seems have some problem I can not run on by board even build finished.

PR #4636 (2017-07-07): Double escape defines in gnuarmeclipse export
Fixes a bug where quoting gets stripped by the shell used in the makefile
and another bug where the lack of escaping would cause parser errors in
eclipse.

Resolves #4622