ARMmbed/mbed-os #4624: Eclipse Exporters broken

awneil, 2017-06-23:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug

- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
XDOT_L151CC

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]

**mbed-cli version:** 
1.1.1

**Expected behavior**
Export should create fully working Eclipse projects

**Actual behavior**
The makefile export configures incorrect include paths; so, although the build works, the indexer, source browsing, etc, don't.
See: https://developer.mbed.org/forum/bugs-suggestions/topic/27892/

The CDT managed export has an incorrect linker command, so the code does not build (compiles OK; Link fails)
See: https://developer.mbed.org/forum/bugs-suggestions/topic/27891/

**Steps to reproduce**
Export from mbed;
Import into Eclipse.
----------------------------------------------------------------

theotherjimmy, 2017-06-23:
Major stuff:
 1) What example or project are you using? (I chase
 1) What version of mbed-os are you using?
 2) When I export to a managed CDT project using the gnuarmeclipse exporter, I get a section on linker flags like:
```xml
<option id="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags.139833293" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags" valueType="stringList">
  <listOptionValue builtIn="false" value="--wrap=main"/>
  <listOptionValue builtIn="false" value="--wrap=_malloc_r"/>
  <listOptionValue builtIn="false" value="--wrap=_free_r"/>
  <listOptionValue builtIn="false" value="--wrap=_realloc_r"/>
  <listOptionValue builtIn="false" value="--wrap=_memalign_r"/>
  <listOptionValue builtIn="false" value="--wrap=_calloc_r"/>
  <listOptionValue builtIn="false" value="--wrap=exit"/>
  <listOptionValue builtIn="false" value="--wrap=atexit"/>
</option>

```
I verified that these arguements are accepted by the linker (warning _really_ long line):
```
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin -funsigned-char -fno-delete-null-pointer-checks -fomit-frame-pointer -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers  -g3 -T "/home/jimbri01/src/armmbed/mbed-os-example-fat-filesystem/Debug/linker-script-debug.ld" -Xlinker --gc-sections -Xlinker --wrap=main -Xlinker --wrap=_malloc_r -Xlinker --wrap=_free_r -Xlinker --wrap=_realloc_r -Xlinker --wrap=_memalign_r -Xlinker --wrap=_calloc_r -Xlinker --wrap=exit -Xlinker --wrap=atexit -Wl,-Map,"mbed-os-example-fat-filesystem.map" -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -Wl,-n -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group -o "mbed-os-example-fat-filesystem.elf"  ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_adc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_adc_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_comp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_cortex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_crc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_cryp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_cryp_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_dac.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_dac_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_dma.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_flash.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_flash_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_flash_ramfunc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_gpio.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_i2c.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_i2s.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_irda.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_iwdg.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_lcd.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_nor.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_opamp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_opamp_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pcd.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pcd_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pwr.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pwr_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rcc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rcc_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rtc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rtc_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_sd.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_smartcard.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_spi.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_spi_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_sram.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_tim.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_tim_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_uart.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_usart.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_wwdg.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_adc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_comp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_crc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_dac.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_dma.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_exti.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_fsmc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_gpio.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_i2c.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_opamp.o ./mbed-os/targets/TARGET_STM/TARGET [... cut ...]
unused variable 'tmpval' [-Wunused-variable]                                                  | serial_device.c                | /mbed-os-example-fat-filesystem/mbed-os/targets/TARGET_STM/TARGET_STM32L1                           | line 207  | C/C++ Problem
this is the location of the previous definition                                               | stm32l1xx_hal_conf.h           | /mbed-os-example-fat-filesystem/mbed-os/targets/TARGET_STM/TARGET_STM32L1/device                    | line 90   | C/C++ Problem

(translated to markdown instead of taking a screenshot)

theotherjimmy, 2017-06-23:
Also, there is no way we can know where your system includes are at export time on the website.

awneil, 2017-06-23:
1. I am using the **Dot-Examples** from: 

https://developer.mbed.org/platforms/MTS-xDot-L151CC/#example-programs

2.  mbed-os version 5.4.2:
```
#define MBED_LIBRARY_VERSION 139

#if MBED_CONF_RTOS_PRESENT
// RTOS present, this is valid only for mbed OS 5
#define MBED_MAJOR_VERSION 5
#define MBED_MINOR_VERSION 4
#define MBED_PATCH_VERSION 2
```

3. In the **.cproject** file for the CDT Project:
```
<option id="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags.799043212" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags" valueType="stringList">
	<listOptionValue builtIn="false" value="--wrap=main"/>
	<listOptionValue builtIn="false" value="--wrap=_malloc_r"/>
	<listOptionValue builtIn="false" value="--wrap=_free_r"/>
	<listOptionValue builtIn="false" value="--wrap=_realloc_r"/>
	<listOptionValue builtIn="false" value="--wrap=_calloc_r"/>
	<listOptionValue builtIn="false" value="--wrap=exit"/>
	<listOptionValue builtIn="false" value="--wrap=atexit"/>
</option>
```

Splitting your command line so the options are all visible:
```
arm-none-eabi-g++ 
-mcpu=cortex-m3 
-mthumb 
-O0 
-fmessage-length=0 
-ffunction-sections 
-fdata-sections 
-fno-builtin 
-funsigned-char 
-fno-delete-null-pointer-checks 
-fomit-frame-pointer 
-Wall 
-Wextra 
-Wno-unused-parameter 
-Wno-missing-field-initializers  
-g3 
-T "/home/jimbri01/src/armmbed/mbed-os-example-fat-filesystem/Debug/linker-script-debug.ld" 
-Xlinker --gc-sections 
-Xlinker --wrap=main 
-Xlinker --wrap=_malloc_r 
-Xlinker --wrap=_free_r 
-Xlinker --wrap=_realloc_r 
-Xlinker --wrap=_memalign_r 
-Xlinker --wrap=_calloc_r 
-Xlinker --wrap=exit 
-Xlinker --wrap=atexit 
-Wl,-Map,"mbed-os-example-fat-filesystem.map" 
-DMBED_DEBUG 
-DMBED_TRAP_ERRORS_ENABLED=1 
-Wl,-n 
-Wl,--start-group 
-lstdc++ 
-lsupc++ 
-lm 
-lc 
-lgcc 
-lnosys 
-Wl,--end-group 
-o "mbed-os-example-fat-filesystem.elf"  ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_adc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_adc_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_comp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_cortex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_crc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_cryp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_cryp_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_dac.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_dac_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_dma.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_flash.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_flash_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_flash_ramfunc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_gpio.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_i2c.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_i2s.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_irda.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_iwdg.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_lcd.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_nor.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_opamp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_opamp_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pcd.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pcd_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pwr.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_pwr_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rcc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rcc_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rtc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_rtc_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_sd.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_smartcard.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_spi.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_spi_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_sram.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_tim.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_tim_ex.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_uart.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_usart.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_hal_wwdg.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_adc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_comp.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_crc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_dac.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_dma.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_exti.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_fsmc.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_gpio.o ./mbed-os/targets/TARGET_STM/TARGET_STM32L1/device/stm32l1xx_ll_i2c [... cut ...] in.o   
-lstdc++ 
-lsupc++ 
-lm 
-lc 
-lgcc 
-lnosys
Finished building target: mbed-os-example-fat-filesystem.elf
```

And there we see the important difference!

In your command line, the `--wrap=` options are all prefixed by `-Xlinker` - which means that they are to be passed through to the Linker. This is an alternative to `-Wl`.

The problem in my case is that neither `-Xlinker` nor `-Wl` is used; so they are given to the compiler - where they are, indeed, invalid and, thus, cause the error.

theotherjimmy, 2017-06-23:
@awneil Agreed. I wonder if the version of GNU ARM Eclipse is the key to that difference?

I'm using version 2.2.1.201701141320

awneil, 2017-06-23:
> there is no way we can know where your system includes are at export time on the website

That is a fair point.

It should, then, be mentioned in the notes.

theotherjimmy, 2017-06-23:
> It should, then, be mentioned in the notes.

Definitely. I'll put in a note to write it up and mention you on the Handbook PR.

awneil, 2017-06-23:
Thanks.

They way I worked them out was to look at the CDT project...

theotherjimmy, 2017-06-23:
Really, they should be included by GCC in the search path by default... 

If not then a simple
```bash
 arm-none-eabi-gcc --print-search-dirs
```
should work

awneil, 2017-06-23:
Which particular GNU ARM Eclipse plugin are you looking at?

I have:
[image]

_(is there a way to extract that as text?)_

awneil, 2017-06-23:
> they should be included by GCC in the search path by default

Yes, GCC itself is fine - it's telling the Eclipse indexer that's the problem!

(this is just for the makefile project - CDT sorts it out itself)

theotherjimmy, 2017-06-23:
That looks pretty old. If you can update your version of eclipse and GNU ARM Eclipse that might help.

> GCC itself is fine - it's telling the Eclipse indexer that's the problem!

That makes sense, and also might be sorted out by upgrading.

theotherjimmy, 2017-06-23:
I'll make sure to put all of this information into the appropriate section in the Handbook. I will request your review when it's done. After that, we can close this issue, supposing that there are no outstanding things to be fixed.

awneil, 2017-06-23:
If I hack the CDT **.cproject** file thus:
```
<option id="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags.799043212" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags" valueType="stringList">
	<listOptionValue builtIn="false" value="-Wl,--wrap=main"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_malloc_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_free_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_realloc_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_calloc_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=exit"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=atexit"/>
</option>
```

ie, manually add the `-Wl,` prefix 

It builds!

theotherjimmy, 2017-06-23:
Awesome! I'll make sure to put that into the docs in a bit. I'm working on compiling all of these workarounds right now.

awneil, 2017-06-23:
The hack would be just a workaround - I'll try the update(s) ...

theotherjimmy, 2017-06-23:
Agreed, I would mention that in the Docs.

chrissnow, 2017-06-23:
export to Keil is not perfect either, the paths all end up one level too shallow (all the files have .. instead of .)
easy work around is to stick the UVPROV inside a folder, been meaning to fix it and PR.

It's odd because I only have this problem with the xDot (I think)

theotherjimmy, 2017-06-23:
@chrissnow Thanks for the information. That info indicates that there might be a bug here too.

awneil, 2017-06-23:
> the paths all end up one level too shallow

Hmmm .... that's exactly what I had with the Eclipse makefile export - so maybe the bug is somewhere "before" the IDE-specific bit ... ?

theotherjimmy, 2017-06-23:
>   so maybe the bug is somewhere "before" the IDE-specific bit ...

That's my intuition as well.

theotherjimmy, 2017-06-23:
@awneil @chrissnow Could you tell me if you exported with mbed CLI or the developer website? This bug might only affect one of those two environments.

chrissnow, 2017-06-23:
@theotherjimmy  CLI for me.

awneil, 2017-06-23:
I have found it almost impossible to export from the website:

[image]

I _think_ the makefile export was from the website; CDT from the CLI.

awneil, 2017-06-23:
@chrissnow said:
>  I only have this problem with the xDot (I think)

Interesting - it seems that @theotherjimmy also escaped the problem by _**not**_ using an xDot project ...

theotherjimmy, 2017-06-23:
I used the target xDot, but not the Dot-Examples

theotherjimmy, 2017-06-23:
I'm going to try the Dot-Exmaples to see if something changes when I use that example.

theotherjimmy, 2017-06-23:
Nothing out of the ordinary here. The include paths seem correct. It does not build because it can't find mDot.h. I don't think the file exists.... `find -iname "mDot.h"` produces no results.

theotherjimmy, 2017-06-23:
Doh. I forgot to add the libDot-mbed5. :man_facepalming:

theotherjimmy, 2017-06-23:
After importing that, it looks like it's correctly identifying errors.

chrissnow, 2017-06-23:
@theotherjimmy  this might be a thing of the past, 5.4.7 works ok for me.

@awneil I'm sure form your email you were using the v3 dot lib so 5.4.7? yet above it shows 5.4.2?

theotherjimmy, 2017-06-23:
@chrissnow That's good to know. I can't backport fixes to 5.4.2, so updating to 5.4.7 is your best bet here.

chrissnow, 2017-06-23:
@theotherjimmy  it's not that simple with the dot libs, they include an archive that was built against a mbed version, not matching has mixed results.

the latest stable is built against 5.4.2 and dev 5.4.7

theotherjimmy, 2017-06-23:
@chrissnow I noticed. You could always copy the tools folder from a newer version of mbed-os and replace the tools folder in the one you currently have. 

An aside, it looks like the stable one does not have ChannelPlan.h, and therefore does not work with the examples.

theotherjimmy, 2017-06-23:
@chrissnow Where do we stand on this issue? is it still a problem?

awneil, 2017-06-23:
> it looks like the stable one does not have ChannelPlan.h, and therefore does not work with the examples.

Indeed.

Have already moaned to MultiTech about that.

They do note it here: https://developer.mbed.org/teams/MultiTech/code/Dot-Examples/

chrissnow, 2017-06-23:
@theotherjimmy uvision5 export seems fine in 5.4.7. I've not tried eclipse, I know 5.4.2 was broken, and my recent work has been master or 5.4.7 which has exported ok.

@awneil  so far the dev V3 lib has been working for me so you might want to use that.
do bear in mind #4576 though...

theotherjimmy, 2017-06-23:
Thanks @chrissnow. I'm going to wait for @awneil's feedback on the status of Eclipse in 5.4.7 before closing this issue.

0xc0170, 2017-06-26:
cc @mfiore02 (see https://github.com/ARMmbed/mbed-os/issues/4624#issuecomment-310756384, and https://github.com/ARMmbed/mbed-os/issues/4624#issuecomment-310754332)

awneil, 2017-06-26:
I needed to keep the old setup for another project - didn't want to break that.
So did a new install of Eclipse Neon3 & gnuarmeclipse plugins v3.4.1-201704251808

The CDT project imported & built successfully - the linker options are now correctly prefixed with -Xlinker:
[image]

The makefile project still has the same Path issues.

awneil, 2017-06-26:
I kept the same GCC toolchain because, last time I looked, the mbed documentation insisted that 4.9 should be used, and 5 should **not**.

But when I look today, it says:

> Use version 6 of GCC ARM Embedded; version 5.0 or any older version might be incompatible with the tools.

?!?!

(but, I guess, that's a separate issue)

0xc0170, 2017-06-26:
@awneil It was updated for 5.5 release, supportss gcc arm 6.x

awneil, 2017-06-26:
@0xc0170 Is there a "history" or "compatibility matrix" of that somewhere?

As @chrissnow said in https://github.com/ARMmbed/mbed-os/issues/4624#issuecomment-310754332 - it's not so simple as just using "latest" ...

(I guess it's something that MultiTech need to document).

theotherjimmy, 2017-06-26:
@awneil It looks like the issue you reported for the gnuarmeclipse exporter was related to the old version of that plugin you were using and the issue you reported with the eclipse_gcc_arm exporter was fixed after version 5.4.2. As far as I can tell, no fixes are needed right now. Therefore I'm closing this issue.

I took a look at the documentation on docs.mbed.org, and you're correct: the compiler versions seem to have changed for the 5.4 release series. That's not good. @AnotherButler is there something we can do about this?

closed 2017-06-26 by theotherjimmy

mfiore02, 2017-06-26:
Mike at MultiTech here. Because we release our lora stack for mDot and xDot precompiled, exporting gets pretty tricky. Like @chrissnow mentioned, even using a different mbed-os version between the app and the library can cause issues in the application. I've found it easier to set up a custom build command in your IDE environment that just uses mbed-cli in the background instead of actually trying to export the project.

@awneil I did put some temporary updates in our doc discussing the current differences between our dev and production builds. I updated our dev builds to 3.0.0 release candidate so that our customers could start playing with new features. The dev builds will be updated soon - when we actually release 3.0.0. Sorry for the confusion there.

Side note about compiler version. Our Dot Libraries are currently being built with GCC 4.9, so I think the app will need to be compiled with that version as well.

theotherjimmy, 2017-06-26:
>  Our Dot Libraries are currently being built with GCC 4.9, so I think the app will need to be compiled with that version as well.

That's correct

@mfiore02 You should update to GCC 6 when you update to mbed os >= 5.5.0

mfiore02, 2017-06-26:
@theotherjimmy Agreed. Probably our next release. I think this one will end up staying with 5.4.7 as it's too late in the cycle for a change like that.