ARMmbed/mbed-os #4626: Clarify the purpose of acquire in SPI & discuss its usage in SD driver

deepikabhavnani, 2017-06-23:
## Question
At present SD device driver uses SPI lock and unlock. In current sample since we are using only single SPI device (SD), it should work fine. But in case of mutliple SPI devices, we should be setting frequency and format as well before (Correct me if I am wrong).

If my understanding above is correct, I should be using acquire and than lock SD device.
Current implementation of SPI has unnecessaru multiple locking for these API's which may affect the performance. Wanted to understand the reason behind code flow as implemented in SPI [https://github.com/ARMmbed/mbed-os/blob/master/drivers/SPI.cpp]

format -> lock -> (Make owner == NULL) -> acquire -> lock -> set format + frequency -> unlock -> unlock
Since we are locking and explicitly setting current SPI object as owner below solution shall work fine. 
format -> lock ->  set format -> unlock 

Let me know if I am missing something in this.

@sg- @geky @c1728p9

c1728p9, 2017-06-23:
All the public functions are locked to ensure thread safety, even when used with legacy code which doesn't explicitly lock.

I agree there is some unnecessary duplicate locking with the call to **acquire**. To increase performance this could be fixed by creating a private **_aquire** function which doesn't use locking. All the places where **aquire** is called with the lock already held could be replaced with **_aquire** for a speed increase.

Finally, you don't need an explicit call to acquire, as this is done internally.

0xc0170, 2017-06-26:
@deepikabhavnani Does the above clarify the SPI locking ?

deepikabhavnani, 2017-06-26:
@0xc0170 yes it is clear, shall I implement private _acquire() function and replace it in SPI code, to remove multiple locks?

Also for format and frequency functions, we can update the format or frequency which is required by directly calling respective functions, instead of using call to acquire. Does that sound correct?

closed 2017-07-11 by deepikabhavnani