ARMmbed/mbed-os #4627: wait function causes system hang on

shayo, 2017-06-24:
- Type: Bug
- Priority: Major 
---------------------------------------------------------------
## Bug
Calling "wait" causes system to hang. Minimal example that does not work is blinky.

**Target**
NRF52_DK (custom board running on nrf52832)

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
0.9.10

**Expected behavior**
a delay...

**Actual behavior**
system hangs.

**Steps to reproduce**

compile and upload blinky.

This is not a custom board issue since using an older version of mbed os works like a charm.
also, us_ticker_read seems to be functioning correctly and delays are working when defined like this:
while (us_ticker_read() -t0 < XXX);

shayo, 2017-06-24:
I was able to trace this to changes that were introduced between rev127 and rev128.
wait() works in rev127, but causes system hang in rev128.
The main difference was introduction of RTOS, which changed the implementation of wait_us.
In rev127, it simply pulls on us_ticker_read().
In rev128, it calls  Thread::wait((uint32_t)ms), which is causing the system to hang.

Can someone reproduce this on NRF52DK?
More importantly right now, since I am not using RTOS functionality at the moment, is there a flag to disable it and use the non-rtos implementation?

0xc0170, 2017-06-26:
cc @pan- @nvlsianpu @anangl

akrnvm, 2017-06-28:
hi,
i have facing slimier problem with my custom board.
In mbed OS 5.4.7 all test-cases are passing.but from mbed 5.5.0 onwards  timer test cases (with RTOS) are failing.
my device is running in low frequency (18MHZ).
In low frequency, the wait_us calls "Thread::wait((uint32_t)ms)", its taking extra micro seconds to execute.
due to extra 0.33 uS ,test-cases are failing.
i have verified my us_ticker code with high frequency(40MHz) custom board and its working fine.
if i have commenting "Thread::wait((uint32_t)ms)", test case is passing
is any solution to avoid this RTX_delay handling timing?

pan-, 2017-07-11:
@akrnvm @shayo Could you test if #4736 fixes your issue ?

shayo, 2017-07-11:
Yes, I'll try it.
Is there an easy option to clone the master branch with this pull request?
(sorry, I'm not very git savvy).

pan-, 2017-07-11:
@shayo please see this [document](https://help.github.com/articles/checking-out-pull-requests-locally/). 

```bash
git fetch origin pull/4736/head:test_nrf52_stack_fix
git checkout test_nrf52_stack_fix
```

shayo, 2017-07-12:
I tried to check out the latest from the master branch, then I applied the pull.
After uploading to the device, it did not output anything, so I suspect things are still not working properly. I did not investigate further.

jrobeson, 2017-11-29:
@shayo  Can you try again? this is probably fixed now.

shayo, 2017-12-06:
Seems to be working now,

closed 2017-12-06 by shayo