ARMmbed/mbed-os #4630: RTX: release-candidate is missing mutex fix

adamgreen, 2017-06-25:
I hit an issue today where RTX would enter an infinite loop when it attempted to release a previously acquired osMutexPrioInherit mutex.

A fix for this issue has already been made in the ARM-software/CMSIS_5 repository but doesn't appear to have made it into the release-candidate mbed-os branch yet.
https://github.com/ARM-software/CMSIS_5/commit/729f6ab08540342b7de7612fac103539d5b2f168

0xc0170, 2017-06-26:
cc @bulislaw

adamgreen, 2017-06-26:
@0xc0170 let me know that 5.5 had actually released. I had missed that. I verified that this patch is still required for the 5.5 branch. https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.5/rtos/rtx5/TARGET_CORTEX_M/rtx_mutex.c#L190

bulislaw, 2017-06-26:
We haven't updated CMSIS_5 for some time now. We need to have a think how we will update CMSIS/RTX, historically we didn't do that very often. Maybe at least we can port bug fixes over.

PR #4646 (2017-06-29): RTX: Fixed RTXv5 mutex owner list handling.
The prev-pointer of the running threads mutex list was not set when
multiple mutexes are acquired at a time. This leads to a corrupted list
if the mutexes are not freed in reversed order.

Original commit for CMSIS_5:
commit 729f6ab08540342b7de7612fac103539d5b2f168
Author: Jonatan Antoni <jonatan.antoni@arm.com>
Date:   Wed Apr 19 15:38:43 2017 +0200

Fix ported from CMSIS_5 repo to address #4630 
## Status
**READY**

Close #4630 

It's a point fix brought over from CMSIS repo, but we should still think about having some process to update CMSIS in re [... cut]
closed 2017-06-29 by theotherjimmy