ARMmbed/mbed-os #4637: exporting with eclicpse exporters create 'not a CDT project'

JojoS62, 2017-06-26:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32072RB and other

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.0.0

**meed-os sha:**
a969883

**Expected behavior**
imported project generated by gnuarmeclipse should compile without errors

**Actual behavior**
compile option disabled, project is not a valid CDT project (shown in project settings)

**Steps to reproduce**
create simple project, export with gnuarmeclipse, import to eclipse IDE.
Check project settings, no compiler settings availible 

The problem is a preprocessor setting with a string as value:

[image]

This setting is generated by the exporter as:
<listOptionValue builtIn="false" value="CMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h""/>

but the quotes arround cmsis_nvic.h are not read correctly and the project file becomes invalid. 
The exporter should create this line as:

```
<listOptionValue` builtIn="false" value="CMSIS_VECTAB_VIRTUAL_HEADER_FILE=&quot;cmsis_nvic.h&quot;"/>
```

JojoS62, 2017-06-26:
fix in progress with PR #4636

closed 2017-06-26 by JojoS62

chrylis, 2017-10-05:
Thanks for explaining the problem clearly; I was able to manually edit my `.cproject` and get things working. Do you know when the CLI will have the fix applied?