ARMmbed/mbed-os #4643: Adding RTL8711, RTL8710

malachib, 2017-06-27:
### Description
- Type: Question/Enhancement
- Related issue: #4438 

-----------------------------------------------------------------

## Question

I am excited to see Ameba support officially materializing in mbed-os!
Any chance the RTL871x series are invited to the party?

0xc0170, 2017-06-27:
cc @Archcady

Archcady, 2017-06-28:
Hi @malachib, thank you for your interest. The latest mbed-os 5.5 does not support our RTL871x series. And for future support, it still need some research before any decision is made.

malachib, 2017-06-28:
OK, thank you for the response.  I hope RTL871x does get targeted, as an inexpensive & easy WiFi SoC mbed OS solution would be pretty sweet

xiangzhai, 2017-11-22:
Hi @Archcady thanks for your kind response! [mbed-os-example-wifi](https://github.com/ARMmbed/mbed-os-example-wifi) for RTL8195A target is including Realtek SDK, [lib_wlan_mbed_gcc.a](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/TARGET_RTL8195A/device/TOOLCHAIN_GCC_ARM/lib_wlan_mbed_gcc.a) the static library's size is weird... #5517, mbed CMSIS RTOS wrapper, HAL, lwip, mbedtls, netsocket, filesystem, mbed rtos, drivers, platform, and events. If build with [GCC ARM Embedded Toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm), the binary size is 367KB, and ELF size:

```
   text    data     bss     dec     hex filename
 275232   54704  473484  803420   c425c mbed-os-example-wifi.elf
```

The binary size is OK for RTL8195A, but might too big for [RTL8710BN](http://www.realtek.com.tw/products/productsView.aspx?Langid=1&PFid=45&Level=4&Conn=3&ProdID=361). I argue that: it might doesn't need `freertos_intfs.o`, `freertos_isr.o`, `freertos_recv.o`, `freertos_xmit.o`, `freertos_ioctl.o`, `freertos_mp.o` or `freertos_skbuff.o` for mbed-os? Realtek SDK [supports](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/sdk/os/os_dep/include/osdep_service.h#L103) freertos, ecos and mbed-os, but perhaps it might be able to delete useless object files only for mbed? And there are `.debug_str`, `.debug_abbrev`, `.debug_info`, `.debug_frame`, `.debug_line` sections in the object files, so perhaps it might be able to turn off `-g` debug and open `-O3` optimization? Please give me some hint about how to tailor, @0xc0170 and please how to make mbed-os smaller :) thanks a lot!

Regards,
Leslie Zhai - a LLVM developer https://reviews.llvm.org/p/xiangzhai/

0xc0170, 2018-07-03:
Thanks for your interest for these platforms. There is currently no plan to add support. @Archcady please correct me if it is.

Feel free to add these as new targets via new pull requesdt. I'll close the issue as won't fix.

closed 2018-07-03 by 0xc0170

janjongboom, 2019-03-20:
FYI https://github.com/ARMmbed/mbed-os/pull/7965