ARMmbed/mbed-os #4651: REALTEK_RTL8195AM compiles, but fails to flash in uvision

MarceloSalazar, 2017-06-27:
REALTEK_RTL8195AM compiles in uvision, but fails to find the correct target to flash to.

mbed OS: `92fbf2a9b3988d430482fc25a6077f2462e2a634` (mbed OS 5.5)
mbed CLI version: `1.1.1`

```
".\BUILD\mbed-os-example-blinky.axf" - 0 Error(s), 297 Warning(s).
Build Time Elapsed:  00:01:38
Load "C:\\cygwin64\\home\\micray01\\mbed-os\\mbed-os-example-blinky\\BUILD\\mbed-os-example-blinky.axf" 
No Algorithm found for: 10006000H - 1000FFFFH
No Algorithm found for: 10010000H - 10016F83H
Erase skipped!
Error: Flash Download failed  -  "Cortex-M3"
Flash Load finished at 17:34:35
```

Issue original raised here: https://github.com/ARMmbed/mbed-os-example-blinky/issues/62

@mray19027 @0xc0170

theotherjimmy, 2017-06-27:
IIRC, This target did not have a device_name set in targets.json, because we did not have a CMSIS PACK for it. That's required to flash.

0xc0170, 2017-06-28:
@Archcady Once this target gets cmsis pack support, please update this thread

MarceloSalazar, 2017-09-14:
@Archcady any update on this?

Archcady, 2017-09-15:
Hi @MarceloSalazar, sorry we still don't have cmsis pack support for now. I'll update this when we have the support. Thanks for your concern.

samchuarm, 2017-12-14:
Hi @Archcady can you plz confirm if this is resolved now?

Archcady, 2017-12-15:
Hi, Is this being built in the Keil IDE or just using the ARMCC compiler inside the Keil MDK package?

Archcady, 2017-12-18:
When I try to export to uvision, the cli says that REALTEK_RTL8195AM is not supported by uvision. Is there a way to resolve this??

MarceloSalazar, 2017-12-18:
Exporter issue reported here: https://github.com/ARMmbed/mbed-os-example-blinky/issues/104

0xc0170, 2017-12-18:
> When I try to export to uvision, the cli says that REALTEK_RTL8195AM is not supported by uvision. Is there a way to resolve this??

It needs `device_name` in targets.json file: https://os.mbed.com/docs/v5.6/tools/adding-and-configuring-targets.html

theotherjimmy, 2017-12-18:
@0xc0170 @Archcady You also have a post build script for your target. You may have to add an entry into the post build script whitelist [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/uvision/__init__.py#L123-L131). Please verify that uVision is capable of replicating the process in your post build script before adding it.

Archcady, 2017-12-19:
In which .pdsc file can I find the device_name for RTL8195AM, does it already exist in index.json?

Archcady, 2017-12-19:
I have been looking into the issue related to the exporter, I also went through the documentation given on the ARM website: https://os.mbed.com/docs/v5.6/tools/adding-and-configuring-targets.html
From what I understand the “device_name” property must match with the data that is parsed by the ARmPackManager from the CMSIS packs target information. It is known that our RTL8195AM does not have any CMSIS pack information so how exactly do I assign the “device_name” for our target?

samchuarm, 2018-09-10:
Hi @ARMmbed/team-realtek , can you please confirm if this is still not supported on RTL8195AM?

samchuarm, 2018-09-11:
Hi @M-ichae-l , may we know why it cannot be supported on RTL8195AM? And do you plan to support it soon? Thanks

M-ichae-l, 2018-09-11:
@samchuarm 
RTL8195AM does not support download with uvision yet.

samchuarm, 2018-09-11:
@M-ichae-l , do you plan to have CMSIS PACK support for this target?

M-ichae-l, 2018-10-02:
Hi, @samchuarm ,
We don't have the plan to support this feature currently.

samchuarm, 2018-10-03:
@M-ichae-l , the reason you can't support download with uVision is because you don't have CMSIS-Pack for RTL8195AM target? Then can you confirm is there a plan to get CMSIS-Pack support for this target?

Reference documentations:
https://os.mbed.com/docs/latest/porting/porting-exporter-targets.html
https://os.mbed.com/docs/v5.10/tools/adding-and-configuring-targets.html
http://www.keil.com/pack/doc/CMSIS/Pack/html/index.html

M-ichae-l, 2018-10-05:
@samchuarm 
I have confirmed that no plan on CMSIS-Pack support currently.

deepikabhavnani, 2018-12-28:
> I have confirmed that no plan on CMSIS-Pack support currently. We can close this PR.

Closing this

closed 2018-12-28 by deepikabhavnani