ARMmbed/mbed-os #4654: mbed 2 with mbed-rtos build broken

toyowata, 2017-06-28:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F, LPC1768

**Toolchain:**
ARM (online IDE)

**Toolchain version:**
5.06

**Expected behavior**
rtos example build successful

**Actual behavior**
RTOS example can not build (compile error)

**Steps to reproduce**

* Import rtos_basic example:
https://developer.mbed.org/compiler/#import:/users/mbed_official/code/rtos_basic/

* Set target to LPC1768 (or K64F)
* Update mbed library to 145
* Build example and I got error below:

```
Error: #error directive: "no target defined" in "mbed-rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c", Line: 54, Col: 4
```

which is...
```
#ifndef OS_TASKCNT
 #error "no target defined"
#endif
```

I could not see any problem in mbed library rev 144 or older.

0xc0170, 2017-06-28:
@toyowata Thanks for reporting, will have a look shortly at this one

0xc0170, 2017-06-28:
I can reproduce this, will send a patch soon

0xc0170, 2017-06-28:
Pull request proposed with a patch #4657