ARMmbed/mbed-os #4658: mbed_blinky build broken in some BLE targets

toyowata, 2017-06-28:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
nRF51-DK, nRF51-Dongle and TY51822R3

**Toolchain:**
ARM (online IDE)

**Toolchain version:**
5.06

**Expected behavior**
mbed_blinky example build successful

**Actual behavior**
mbed_blinky example can not build (compile error)

**Steps to reproduce**

* Import mbed_blinky example:
https://developer.mbed.org/teams/mbed/code/mbed_blinky/

* Set target to nRF51-DK
* Update mbed library to 145
* Build example and I got error below:

```
Error: Symbol SysTick_Handler multiply defined (by us_ticker.o and startup_nRF51822.o).
```

I could not see any problem in mbed library rev 144 or older.

0xc0170, 2017-06-28:
cc @pan- @nvlsianpu

pan-, 2017-06-29:
It works well if compiled locally from the sources but doesn't in the online IDE. 
I suspect that it is related to how ARM compiler handle weak symbols when they are in a library.

0xc0170, 2017-06-29:
we shall not use weak IRQ handlers within HAL, we got NVIC set/get vectors so use that.

pan-, 2017-06-29:
@bulislaw @nvlsianpu Any reason to keep the weak definition of `Systick_Handler` in `us_ticker.c` ([see](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/us_ticker.c#L306-L319)) ? 

It is only used if the RTOS is present ([see](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/us_ticker.c#L482)) and in that case it is imported from `rtx_os.h`.

nvlsianpu, 2017-06-29:
@pan- No reason to have these weka symbol. Can you remove it? I haven't time for mbed until next week.

pan-, 2017-06-29:
@nvlsianpu Thank you, I'm waiting for @bulislaw answer before submitting the fix, he introduce the change, maybe there is a deeper reason behind the inclusion of this weak symbol.

bulislaw, 2017-06-29:
@pan- I don't think there is one really. I've just translated old symbol name to a new one, it used to be `MBED_WEAK void OS_Tick_Handler() { }`.

toyowata, 2017-07-20:
Confirmed the fix in mbed library rev 147.  Thanks.

closed 2017-07-20 by toyowata