ARMmbed/mbed-os #4663: Odin board fails to connect

sarahmarshy, 2017-06-28:
### Description
- Type: Bug
- Related issue: git fetch origin pull/ID/head:BRANCHNAME
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
 UBLOX_EVK_ODIN_W2

Output of `mbedls`:
```
+-------------------+----------------------+-------------+-------------+--------------------------+-----------------+
| platform_name     | platform_name_unique | mount_point | serial_port | target_id                | daplink_version |
+-------------------+----------------------+-------------+-------------+--------------------------+-----------------+
| UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[0] | D:          | COM9        | 12360221022268373D66EB3C | 0221            |
+-------------------+----------------------+-------------+-------------+--------------------------+-----------------+
```

**Toolchain:**
GCC_ARM

**meed-os sha:**
I have tried the following:
* tip of master
* #4571 rebased onto master

Master should include these supposed fixes:
#4444
#4567
#4543 

**Expected behavior**
[This line]( https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/main.cpp#L112) suceeds.

**Actual behavior**
It fails with -3012 -- device error.

**Steps to reproduce**
Try to run https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/main.cpp#L112 on the Odin board using the current master of mbed-os.

## Question

@andreaslarssonublox is there something else I need to do for this to work? I can see here - https://github.com/ARMmbed/mbed-os/pull/4567 that you said you were not seeing this failure anymore.

andreaslarssonublox, 2017-06-29:
Hi, I just tried it and it worked for me. I did the following:

Clone https://github.com/ARMmbed/mbed-os-example-wifi
mbed update
After this I got SHA-1: 82638173e0feaf551df125fda0e19f85a8e608b1 for mbed-os-example-wifi

Change to latest mbed-os SHA-1: a39ac603052a35d9e474b3d87e4511439d1154a0
Rebase/cherry-pick 3934b30b58076b50d63140ae3ac6069797e8697d from geky:events-use-rtx-event-flags
Change SSID and password in mbed_app.json according to my access point
Build
After copying the file to the board I got:

```
WiFi example

Scan:
Network: cb-jenkins-wpa secured: WPA2 BSSID: 0:14:C2:b7:68:31 RSSI: -66 Ch: 1
.
.
15 networks available.

Connecting...
Success

MAC: d4:ca:6e:70:49:5d
IP: 10.12.71.180
Netmask: 255.255.255.0
Gateway: 10.12.71.254
RSSI: -62

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

Done

```

yennster, 2017-06-29:
@sarahmarshy I just got the Odin board to connect to the `iotlab` wifi. There must have been something wrong with the access point these past few days that got fixed yesterday/this morning? I used the same master branch of [mbed-os-example-wifi](https://github.com/ARMmbed/mbed-os-example-wifi).

```
WiFi example

Scan:
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -77 Ch: 1
Network: ESP_DC84B6 secured: None BSSID: 1A:FE:34:dc:84:b6 RSSI: -92 Ch: 1
2 networks available.

Connecting...
Success

MAC: d4:ca:6e:70:47:9e
IP: 192.168.1.226
Netmask: 255.255.255.0
Gateway: 192.168.1.1
RSSI: -77

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

Done
```

theotherjimmy, 2017-06-29:
So, are we going to close this issue then?

closed 2017-06-30 by sarahmarshy