ARMmbed/mbed-os #4669: feature request: Heap traces

jupe, 2017-06-29:
### Description
- Type: Enhancement
- Related issue: #4573, ...
- Priority: Major

## Enhancement

**Reason to enhance or problem with existing solution**
memory traces are good when analysing fragmentation or who is spending memory in very low level (caller level). But it doesn't tell anything about real heap usage or malloc overheads. memory traces are also very heavy because it prints line to stdout every time when some alloc (`malloc`, `free`, `realloc`,..) is called, so it cannot be activated in all cases, e.g. "end-user" applications which start failing if memory traces takes too much cpu/io time. 
There is already API to fetch heap state but that is application level API - so application needs interaction to get heap state to get printed out.

**Suggested enhancement**
If we could have similar feature than memory-trace, which prints heap state every time when it  changes so application doesn't need to care about it at all. This approach would be also much more light because heap state doesn't change so often compared to malloc calls (memory traces). Feature could be behind compile time flag so it doesn't affect performance if it is not activated.

This should be quite easy to implement in gcc at least. 

**Pros**
allows to track heap consumption using traces - in all applications / tests automatically

**Cons**
none

SenRamakri, 2018-07-02:
Hi @jupe - Please see mbed_stats_heap_get() in https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_alloc_wrappers.cpp#L54. Is this what you are looking for? If we need to print the heap info periodically, we can use an event loop in application level to print this periodically.

jupe, 2018-07-04:
> Is this what you are looking for?

I'm looking for "background printer" for stats_heap

deepikabhavnani, 2018-08-30:
Related : https://github.com/ARMmbed/mbed-os/issues/7343

deepikabhavnani, 2018-08-30:
@jupe -You essentially need combination of memory trace and heaps stats, which is available. User can write there own callback instead of default `mbed_mem_trace_default_callback` and pass it to `mbed_mem_trace_set_callback`. You can use `mbed_stats_heap_get` as a callback.

```
void mbed_mem_trace_user_callback(uint8_t op, void *res, void *caller, ...)
{
    mbed_stats_heap_t heap_stats;
    mbed_stats_heap_get(&heap_stats);
    /** log / send to nw / analyze heap_stats */
}

int main() {
    
    
    mbed_mem_trace_set_callback(mbed_mem_trace_default_callback);
    while (true) {
        void *p = malloc(50);
        wait(0.5);
        free(p);
        mbed_mem_trace_set_callback(mbed_mem_trace_user_callback);
    }
}
```

jupe, 2018-08-31:
Your propose would need modifications to apps, which are not generic solution. Also it would need extra steps and logic to recognize when heap state changes which also increase cpu usage a bit. Original idea was to integrate this feature somehow to heap handler where solution would be most optimized with minimum impact to execution. Current memory tracing is relatively heavy and is not suitable for all cases but this could be suitable for all. If we could switch on heap traces in all tests we have in one shoot we could start collecting metric about how much heap different scenarios needs by free. Does this make senses?

deepikabhavnani, 2018-08-31:
> Also it would need extra steps and logic to recognize when heap state changes which also increase cpu usage a bit. 

On every heap state change (alloc / free) callback is called.

> Original idea was to integrate this feature somehow to heap handler

Please clarify which heap handler are you referring to here?

>  If we could switch on heap traces in all tests we have in one shoot we could start collecting metric about how much heap different scenarios needs by free

Heap trace is enabled in all tests in Mbed OS repo. We get heap stats at the end of each test in CI.

deepikabhavnani, 2018-09-05:
@jupe - Do you have any more concerns with heap trace functionality?

closed 2018-09-17 by deepikabhavnani