ARMmbed/mbed-os #4670: No mbedTLS server support

sherrellbc, 2017-06-29:
### Description
- Type: Enhancement | Question
- Priority: Major
----------------------------------------------------------------

## Enhancement

It does not appear to be possible to host an mbedTLS SSL server on mbed-os. I found [this](https://github.com/ARMmbed/mbedtls/blob/development/programs/ssl/ssl_server.c) example on the mbedTLS Github page, but the code was written assuming a UNIX platform. 
Similarly, there are no server examples over at [mbed-os-example-tls](https://github.com/ARMmbed/mbed-os-example-tls).

After searching around a bit I have found the following:
- [tls.mbed](https://tls.mbed.org/discussions/generic/compiling-problems) 
- [developer.mbed](https://developer.mbed.org/questions/69356/How-the-communication-between-client-ser/)
- [developer.mbed](https://developer.mbed.org/questions/76079/mbedtls-linker-error/)

All pointing to there not being any server support for mbedTLS on mbed-os.

A majority of the functionality necessary for implementing a server seems to be available in [net_sockets.c](https://github.com/ARMmbed/mbedtls/blob/development/library/net_sockets.c). Specifically, of course, [bind](https://github.com/ARMmbed/mbedtls/blob/development/library/net_sockets.c#L184) and [accept](https://github.com/ARMmbed/mbedtls/blob/development/library/net_sockets.c#L296). However, due to the [MBEDTLS_NET_C](https://github.com/ARMmbed/mbedtls/blob/development/library/net_sockets.c#L28) macro being undefined, essentially the entire file is excluded at compile-time. However, according to the [documentation](https://github.com/ARMmbed/mbedtls/blob/development/include/mbedtls/config.h#L2015-L2031) for this macro, it seems server support can be had with a little support from the platform. 

Am I overlooking mbedTLS server support or is it currently unimplemented. If unimplemented, is it planned?

0xc0170, 2017-07-03:
Does https://github.com/ARMmbed/mbed-os-example-tls/issues/98#issuecomment-312018849 answer your question?

sbutcher-arm, 2018-06-19:
Mbed TLS should support being a server on Mbed OS, although we don't have a sample for it. Being a server in an embedded or IOT device isn't always a great choice, as it requires being on all the time (no good for battery powered or low power devices) and could be susceptible to attack by other entities on it's network, as it's always listening.

I'll take this as a feature request, and we'll consider adding an Mbed OS sample application.

sbutcher-arm, 2018-07-25:
I've raised this as [a feature request on the TLS example app](https://github.com/ARMmbed/mbed-os-example-tls/issues/183), so I think this issue can now be closed.

closed 2018-07-26 by 0xc0170