ARMmbed/mbed-os #4675: NUCLEO_F303K8 target not supported in 5.5+

malachib, 2017-06-30:
### Description
- Type: Question
- Priority: Major

-----------------------------------------------------------------

## Question

**It appears NUCLEO_F303K8 target is no longer supported.  Should we expect this target to return?**

bulislaw, 2017-06-30:
Still here on master https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L790

What sort of issues do you have with it?

Nodraak, 2017-06-30:
Nope, the [`release_version`](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L790) is only "2". I tried to add "5" and compile, but it fails.

malachib, 2017-06-30:
@Nodraak interesting, looks like it's been that way in the targets.json file for a long time

However, I have been using it with 5.4.x and I think 5.2.x.

It stopped working when moving to 5.5 - but from the looks of things, it was never supposed to work?  

@bulislaw The error specifically is

`Could not compile for NUCLEO_F303K8: Target does not support mbed OS 5`

0xc0170, 2017-07-03:
> Could not compile for NUCLEO_F303K8: Target does not support mbed OS 5

That error is correct, seems this target has not been enabled for mbed 5.x.

cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-07-03:
Hi
Maybe it can't compile because of the small lib use in targets json file:
    "default_lib": "small",

It has been decided that all small memory targets can not be used for OS5.

malachib, 2017-07-03:
OK.  A little frustrating, this is the 2nd suite I purchased specifically for mbed development which has fallen off the supported list.  I get it though, this is the nature of OSS development...

bulislaw, 2017-07-03:
It was never actually supported by mbed OS 5, the are supported versions of mbed OS are listed on the product page https://developer.mbed.org/platforms/ST-Nucleo-F303K8/ it just happened to work if anything. On the side note, when I add '5' to supported version it compiles fine (at least blinky example), I don't have this board to validate, but maybe there's a chance you can hack the support in.

```
diff --git a/targets/targets.json b/targets/targets.json
index 1b6ef220e..c53e0b298 100644
--- a/targets/targets.json
+++ b/targets/targets.json
@@ -796,7 +796,7 @@
         "detect_code": ["0775"],
         "default_lib": "small",
         "device_has_add": ["ANALOGOUT", "CAN", "LOWPOWERTIMER", "SERIAL_FC"],
-        "release_versions": ["2"],
+        "release_versions": ["2", "5"],
         "device_name": "STM32F303K8"
     },
     "NUCLEO_F303RE": {
```

```
$ mbed compile -m nucleo_f303k8
...
+--------------------+-------+-------+------+
| Module             | .text | .data | .bss |
+--------------------+-------+-------+------+
| Fill               |    62 |     4 |   14 |
| Misc               |  3943 |   120 |   68 |
| drivers            |   118 |     4 |  100 |
| hal                |   518 |     0 |   24 |
| platform           |  1108 |     4 |  270 |
| rtos               |   491 |     4 | 4360 |
| rtos/rtx5          |  7990 |   176 | 1368 |
| targets/TARGET_STM |  6658 |     4 |  752 |
| Subtotals          | 20888 |   316 | 6956 |
+--------------------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 7272 bytes
Total RAM memory (data + bss + heap + stack): 7272 bytes
Total Flash memory (text + data + misc): 21204 bytes

Image: ./BUILD/nucleo_f303k8/GCC_ARM/mbed-os-example-blinky.bin
```

malachib, 2017-07-04:
@bulislaw I'll give it a shot

@jeromecoutant it sounds to me like this means for the foreseeable future, F303K8  won't be officially supported

jeromecoutant, 2017-07-05:
As long as OS2 is supported by ARM MBED, it will be supported...

theotherjimmy, 2017-07-05:
@jeromecoutant @malachib That should probably be phrased as "for the forseeable future, the F303K8 will not be officially supported as an mbed OS 5 target."

malachib, 2017-07-05:
@jeromecoutant @theotherjimmy wish it was in OS 5, but these answers work for me :+1: Thank you all, closing the issue

closed 2017-07-05 by malachib