ARMmbed/mbed-os #4680: wait function causes Greentea test failure

akrnvm, 2017-06-30:
### Description
- Type: Bug 
- Related issue: `#4627' 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom board

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
GCC_ARM 6.0 | ARM 5.2 | IAR 7.8

**mbed-cli version:** 
(`mbed --version-1.1.1`)

**Expected behavior**

1 millisecond delay
**Actual behavior**

1.3 millisecond delay

**Steps to reproduce**

mbed test -t GCC_ARM -m XXXX -n tests-mbed_drivers-wait_us -v

----------------------------------------------------------------
In mbed OS 5.4.7 all test-cases are passing.
But from mbed 5.5.0 onwards timer test cases (with RTOS) are failing.
my device is running in low frequency (18MHZ) and low RAM (16 KB)

My assumption:
In low frequency, the wait_us calls "Thread::wait((uint32_t)ms)", its taking extra micro seconds to execute.
due to extra 0.33 ms ,test-cases are failing.
```

[1498829667.94][CONN][RXD]` >>> Running 1 test cases...
[1498829667.98][CONN][RXD]
[1498829667.98][CONN][INF] found KV pair in stream: {{__testcase_name;Timers: wait_us}}, queued...
[1498829668.02][CONN][RXD] >>> Running case #1: 'Timers: wait_us'...
[1498829668.06][CONN][INF] found KV pair in stream: {{__testcase_start;Timers: wait_us}}, queued...
[1498829669.40][CONN][INF] found KV pair in stream: {{tick;0}}, queued...
[1498829669.44][HTST][INF] tick! 1498829669.405000
[1498829670.74][CONN][INF] found KV pair in stream: {{tick;1}}, queued...
[1498829670.75][HTST][INF] tick! 1498829670.743000
[1498829672.08][CONN][INF] found KV pair in stream: {{tick;2}}, queued...
[1498829672.09][HTST][INF] tick! 1498829672.078000
[1498829673.43][CONN][INF] found KV pair in stream: {{tick;3}}, queued...
[1498829673.46][HTST][INF] tick! 1498829673.434000
[1498829674.78][CONN][INF] found KV pair in stream: {{tick;4}}, queued...
[1498829674.78][HTST][INF] tick! 1498829674.775000
[1498829676.12][CONN][INF] found KV pair in stream: {{tick;5}}, queued...
[1498829676.15][HTST][INF] tick! 1498829676.119000
[1498829677.45][CONN][INF] found KV pair in stream: {{tick;6}}, queued...
[1498829677.49][HTST][INF] tick! 1498829677.448000
[1498829678.80][CONN][INF] found KV pair in stream: {{tick;7}}, queued...
[1498829678.83][HTST][INF] tick! 1498829678.797000
[1498829680.13][CONN][INF] found KV pair in stream: {{tick;8}}, queued...
[1498829680.18][HTST][INF] tick! 1498829680.133000
[1498829681.48][CONN][INF] found KV pair in stream: {{tick;9}}, queued...
[1498829681.52][HTST][INF] tick! 1498829681.484000
[1498829682.83][CONN][INF] found KV pair in stream: {{tick;10}}, queued...
[1498829682.84][HTST][INF] tick! 1498829682.832000
[1498829682.87][CONN][INF] found KV pair in stream: {{__testcase_finish;Timers: wait_us;1;0}}, queued...
[1498829682.89][HTST][INF] test suite run finished after 15.03 sec...
[1498829682.90][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1498829682.90][HTST][INF] CONN exited with code: 0
[1498829682.90][HTST][INF] No events in queue
[1498829682.90][HTST][INF] stopped consuming events
[1498829682.90][HTST][INF] host test result(): False
[1498829682.90][HTST][WRN] missing __exit event from DUT
[1498829682.90][HTST][WRN] missing __exit_event_queue event from host test
[1498829682.90][HTST][INF] calling blocking teardown()
[1498829682.90][HTST][INF] teardown() finished
[1498829682.90][HTST][INF] {{result;failure}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'FAIL'
mbedgt: test on hardware with target id: 701100000000000000000000000000000000000097969904
mbedgt: test suite 'tests-mbed_drivers-wait_us' ...................................................... FAIL in 50.12 sec
        test case: 'Timers: wait_us' ................................................................. OK in 14.81 sec

```

Each tick is increment by 1.3 millisecond.

I have verified my us_ticker code with high frequency(40MHz) custom board and its working fine.
if i have commenting "Thread::wait((uint32_t)ms)" line and test case is passing.

## Question
Is any solution to avoid this RTX_delay execution time?

0xc0170, 2017-06-30:
What has changed between 5.4.x and 5.5 for your target? only RTOS update? 0.33ms is a big delay. if your wait is using us ticker it works, if rtx delay (I assume it is systick in that case) it fails with this drift?

There's PR to update this host drifting script to be more accurate, readable https://github.com/ARMmbed/mbed-os/pull/4599

akrnvm, 2017-06-30:
Yes, only RTOS update. No driver changes,

Same driver is working with different board which has 40 MHz.

Other TImer cases are working fine.

1. tests-mbed_drivers-timeout
2.  tests-mbed_drivers-ticker 

RTOS timer related cases are Failing :
1. tests-mbedmicro-rtos-mbed-basic 
2. tests-mbedmicro-rtos-mbed-timer

in "single test" all testcases are passing including wait_us.

0xc0170, 2017-07-03:
There might be also changes to ticker that were introduced recently. I would advice to run simple ticker test to confirm your ticker is running or debug the failing tests.

What is failing, are those 2 mbed timer based. Is your RTOS timer configured properly? I assume those 2 are using systick, thus would verify if it is setup properly. 
Other two tests (timetout and ticker) are based on the us ticker, that is not failing.

akrnvm, 2017-07-05:
Thank you.
Test cases are passing after changing the systick configuration.

closed 2017-07-05 by akrnvm

pan-, 2017-07-05:
@akrnvm Could you detail a bit how you managed to solve the issue ?

akrnvm, 2017-07-06:
Initially I have updated my system clock value in mbed_RTX.h with OS_CLOCK  variable.
in mbed OS 5.0 onward, OS_CLOCK  variable is not using and the systick is getting updated in rtx_kernel.c
`osRtxInfo.kernel.sys_freq = SystemCoreClock;`