ARMmbed/mbed-os #4681: REALTEK_RTL8195AM - larger image drops RLT8195AM out of USB

JanneKiiskila, 2017-06-30:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/pull/4665
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195A

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)
1.1.1

**meed-os sha:**
(`git log -n1 --oneline`)

Using fork from Archady, see linked issue 4665

**DAPLink version:**
?

**Expected behavior**

Flashing a mbed-os-example-client with client logs enabled should just enable more logs.

**Actual behavior**

Flashing fails, the whole Realtek Ameba boards gets dropped out of USB and device disappears.

**Steps to reproduce**

compile mbed-os-example;
modify mbed_app.json to enable client logs
    https://github.com/ARMmbed/mbed-os-example-client/blob/master/mbed_app.json#L36
```bash
mbed compile -m REALTEK_RTL8195AM
cp BUILD/REALTEK_RTL8195AM/GCC_ARM/mbed-os-example-client.bin /media/jankii01/MBED && sync
```bash

To recover you must reconnect the USB cable completely.

JanneKiiskila, 2017-06-30:
@Archcady @adbridge @marcuschangarm @0xc0170

JanneKiiskila, 2017-06-30:
When trying to re-flash with the non-debug image - it still fails. This is a hard problem for me now.

```
target_flash_erase_sector() : 0x6d000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x6e000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x6f000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x70000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x71000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x72000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x73000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x74000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x75000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x76000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x77000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x78000
set event : MSC_TIMEOUT_SPLIT_FILES_EVENT 
msc_event_timeout 
main : usb disconnected 
```

I can't even flash in the smaller image now, so is the board bricked?

JanneKiiskila, 2017-06-30:
Flashing in the mbed-os-example-wifi gets this one done;

```
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x4c000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x4d000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x4e000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x4f000
CPU clock : *(0x14) = 0x21 
target_flash_erase_sector() : 0x50000
set event : MSC_TIMEOUT_SPLIT_FILES_EVENT 
msc_event_timeout 
main : usb disconnected 
```
So, there are less sectors to flash, but still it fails to start the app.

...

Several retries laters... 
It started working once I pressed the Pdn button on the opposite side of the SMA-connector.

JanneKiiskila, 2017-06-30:
Yep, it requires 2 flashes back/forth these two mbed-os versions, is the boot address now somehow changing between these two?

0xc0170, 2017-07-03:
cc @Archcady

Archcady, 2017-07-03:
The `Image1 Validation Incorrect !!!` message seems like a corrupted image1, I'm trying to reproduce this at my side

JanneKiiskila, 2017-07-03:
The image is created simply by compiling the mbed-os-example-client so that the debug prints are turned on.

`git clone https://github.com/ARMmbed/mbed-os-example-client.git`
`git checkout easy-rtl-wifi`
- to pull in https://github.com/ARMmbed/mbed-os-example-client/pull/268
- take into use the wifi config, modify it to use WIFI_REALTEK -value.
modify the mbed_app.json to enable the tracing (`"mbed-trace.enable": 0` -> `"mbed-trace.enable": 1`)
`mbed compile -m REALTEK_RTL8195AM`

JanneKiiskila, 2017-07-03:
I updated the DAPLINK SW to it and now I can actually get the trace image also to the device. Client handshake however still fails. 

But, the remedy for this particular problems seems to be to update the DAPLINK.
Kudos for @marcuschangarm  for pointing that out to me.

https://developer.mbed.org/platforms/REALTEK-RTL8195AM/

The instructions are a bit confusing, though.

"To update DAP firmware, please follow the steps:
1.	Press and hold the button next to CON2.
2.	Press the button next to CON1.
3.	Release the button you press and hold in step 1.
"

Instruction 3. should be just "Release the button next to CON2 while keeping button next to CON1 held down." I had a bit of trouble understanding that part.

Anyways, one does NOT simply copy over the file -> I get an out of space error. I had to manually DELETE the ´firmware.bin` 1st and after that copy the newer one (renamed to be `firmware.bin`) to the special drive.

So, I think this issue can be closed with some changes to the documentation stressing the importance of updating the DAPLINK before trying to seriously use that board.

closed 2017-07-03 by JanneKiiskila

marcuschangarm, 2017-07-03:
@JanneKiiskila 

Do you also get a 

```
RTL8195A[Driver]: no beacon for a long time, disconnect or roaming
```

after some time?

JanneKiiskila, 2017-07-04:
Yes, I'm seeing that occasionally as well (TCP).

JanneKiiskila, 2017-07-10:
Even with the updated DAPLINK SW I'm seeing sometimes the "USB device malfunctioned" error notifications in Windows, so there's definitely something still going on with the firmware. Is the very latest DAPLINK used?

Archcady, 2017-07-10:
cc @tung7970

tung7970, 2017-07-11:
@JanneKiiskila Regarding the daplink firmware, that should be the latest one, but it's not using the latest daplink codebase. Will work with RTL8195AM team to get the firmware up to date.

TCP is another issue tho.