ARMmbed/mbed-os #4683: REALTEK_RTL8195AM - WiFi implementation not complete?

JanneKiiskila, 2017-06-30:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)
1.1.1

**meed-os sha:**
(`git log -n1 --oneline`)
commit a3a7846817d7eefe59731762f681f53d2fc80da7 (HEAD -> enable_sdram, origin/enable_sdram)
- from the Archady fork 
https://github.com/ARMmbed/mbed-os/pull/4665

Previous commit;
commit a39ac603052a35d9e474b3d87e4511439d1154a0
Merge: 4fc4405 ec72ac0
Author: Jimmy Brisson <theotherjimmy@gmail.com>
Date:   Wed Jun 28 09:31:25 2017 -0500

    Merge pull request #4159 from adustm/STM_sha256_F439ZI
    
    NUCLEO_F439ZI/mbedtls: add SHA256 hw_acceleration

**DAPLink version:**
?

**Expected behavior**

mbed-os-example-client should be able to connect to the mbed Connector.

**Actual behavior**

mbed-os-example-client is not able to connect, absolutely nothing seems to happen.
(Problem now that we can't enable the full logset due to the other issues).

**Steps to reproduce**

Question - has the ASYNC network API been implemented fully to the REALTEK_RLT8195AM?
Please note that mbed-os-example-wifi does not exercise the ASYNC API at all.

JanneKiiskila, 2017-06-30:
```
K
K
K

KIFI is already running

Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (REALTEK)
[EasyConnect] Connecting to WiFi jannek.iki.fi

RTL8195A[Driver]: set ssid [jannek.iki.fi] 

RTL8195A[Driver]: start auth to ec:1d:7f:92:6d:99

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=1)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address f0:03:8c:e7:a9:cd
[EasyConnect] IP address 192.168.1.149

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
```

It doesn't get very far.

0xc0170, 2017-07-03:
cc @Archcady

JanneKiiskila, 2017-07-03:
That K K K  print and "KIFI" is also a bit strange.

Archcady, 2017-07-03:
This "K" message can be switch off by undefine CONFIG_DEBUG_LOG in platform_autoconf.h inside of REALTEK's target directory. However I didn't get `[EasyConnect] Using WiFi (REALTEK)`, instead I have `[EasyConnect] Using WiFi (ESP8266) ` which obviously is my setting of code being incorrect...

JanneKiiskila, 2017-07-03:
Yes, please modify the mbed_app.json connection line -> change WIFI-ESP8266 to WIFI-REALTEK. 
I can push a new commit to the mbed-os-example-client with a pre-defined config file for Realtek.

JanneKiiskila, 2017-07-03:
Remember to also use the .mbedignore in case you have a compilation problem with Atmel RF.

```     
cp configs/eth-wifi-mbedignore ./.mbedignore 
```

JanneKiiskila, 2017-07-03:
Pushed new commit to `mbed-os-example-client` PR https://github.com/ARMmbed/mbed-os-example-client/pull/268.

JanneKiiskila, 2017-07-04:
Ok, now we managed to get it to register ONCE with UDP- seems every run is a bit different and it either hangs or dies. Other issues we see

1) WiFi connection is not stable - after a reboot, even though my hotspot is available - RTL8195 fails to connect to it.

```
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (REALTEK)
[EasyConnect] Connecting to WiFi jannek.iki.fi

RTL8195A[Driver]: set ssid [jannek.iki.fi] 

RTL8195A[Driver]: start auth to ec:1d:7f:92:6d:99

RTL8195A[Driver]: sta recv deauth reason code(1) sta:ec:1d:7f:92:6d:99
failed: -1
[EasyConnect] MAC address f0:03:8c:e7:a9:cd
[EasyConnect] Connection to Network Failed -3004!

Connection to Network Failed - exiting application...

RTL8195A[Driver]: sta recv disassoc reason code(4) sta:ec:1d:7f:92:6d:99
```

2) With TCP I sometimes get a beacon disconnect

```
RTL8195A[Driver]: no beacon for a long time, disconnect or roaming
```

I think that's a bug and it shouldn't happen?

Archcady, 2017-07-04:
I've got log like this, the register didn't success. 

```
<RTL8195A>
=========================================================

ROM Version: 0.2

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003) 

=========================================================

Check boot type form eFuse

SPI Initial

Image1 length: 0x2d64, Image Addr: 0x10000bc8

Image1 Validate OK, Going jump to Image1

Initializing WIFI ...

WIFI initialized

WIFI is already running
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (REALTEK)
[EasyConnect] Connecting to WiFi RealEZ

RTL8195A[Driver]: set ssid [RealEZ] 

RTL8195A[Driver]: start auth to 9c:1c:12:13:cd:01

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=4)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 00:e0:4c:87:00:58
[EasyConnect] IP address 172.25.24.43
[2K[90m[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 3200[0m

[2K[90m[DBG ][mClt]: M2MObject::create_object_instance - id: 0[0m
[2K[90m[DBG ][mClt]: M2MObject::object_instance(inst_id 0)[0m
[2K[90m[DBG ][mClt]: M2MBase::M2Mbase resource name is EMPTY ===========[0m
[2K[90m[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5501)[0m
[2K[90m[DBG ][mClt]: M2MResourceBase::set_value()[0m

[2K[90m[DBG ][mClt]: M2MResourceBase::is_value_changed() -- true[0m
[2K[90m[DBG ][mClt]: M2MResourceBase::report() - level 0[0m
[2K[90m[DBG ][mClt]: M2MResourceInstance::report() - combined level 0[0m
[2K[90m[DBG ][mClt]: M2MResourceBase::report() - mode = 1, is_observable = 1[0m
[2K[90m[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 3201[0m

[2K[90m[DBG ][mClt]: M2MObject::create_object_instance - id: 0[0m

[2K[90m[DBG ][mClt]: M2MObject::object_instance(inst_id 0)[0m
[2K[90m[DBG ][mClt]: M2MBase::M2Mbase resource name is EMPTY ===========[0m

[2K[90m[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5853)[0m
[2K[90m[DBG ][mClt]: M2MResourceBase::set_value()[0m

[2K[90m[DBG ][mClt]: M2MResourceBase::is_value_changed() -- true[0m

[2K[90m[DBG ][mClt]: M2MObjectInstance::create_dynamic_resource(resource_name 5850)[0m
[2K[90m[DBG ][mClt]: M2MInterfaceFactory::create_object : Name : 1000[0m
[... 283 lines cut ...]

[2K[90m[DBG ][mClt]: M2MInterfaceImpl::timer_expired()[0m
[2K[90m[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 6[0m
[2K[90m[DBG ][mClt]: M2MInterfaceImpl::state_engine[0m
[2K[90m[DBG ][mClt]: M2MInterfaceImpl::state_register[0m
[2K[90m[DBG ][mClt]: stop_listening()[0m
[2K[90m[DBG ][mClt]: close_socket() - IN[0m
[2K[90m[DBG ][mClt]: close_socket() - status: 0 OUT[0m
[2K[90m[DBG ][mClt]: resolve_server_address()[0m
[2K[90m[DBG ][mClt]: send_dns_event()[0m
[2K[90m[DBG ][mClt]: M2MConnectionHandlerPimpl::dns_handler - _socket_state = 0[0m
[2K[90m[DBG ][mClt]: close_socket() - IN[0m
[2K[90m[DBG ][mClt]: close_socket() - status: 0 OUT[0m
[2K[90m[DBG ][mClt]: address family: 2[0m
[2K[90m[DBG ][mClt]: IPv4 Address a9:2d:52:12[0m
[2K[90m[DBG ][mClt]: init_socket - IN[0m
[2K[90m[DBG ][mClt]: Interface count: 1[0m
[2K[90m[DBG ][mClt]: Interface name: [0m
[2K[90m[DBG ][mClt]: Interface no: 0[0m
[2K[90m[DBG ][mClt]: init_socket - port 0[0m
[2K[90m[DBG ][mClt]: init_socket - OUT[0m

[2K[90m[DBG ][mClt]: resolve_server_address - Using TCP[0m
simulate button_click, device not registered
```

JanneKiiskila, 2017-07-04:
Hi @Archcady  - you seem to have some connection problem, it's not able to connect to the COAP server.

Can you reach this address `coap://api.connector.mbed.com:5684`?

How large are the buffers in your WiFi stack? The certificate downloads are getting 2381 bytes a chunk, it seems to repeat that. Also the TLS handshaking is really slow, which might cause problems as well. Is the SSL acceleration enabled in the Ameba board with mbed OS 5.5?

Archcady, 2017-07-04:
For wifi poor connection and no beacon behaviors, we think the reason is probably poor signal if you didn't attach antenna to board. 
How do I check if I can reach the address (`coap://api.connector.mbed.com:5684`) or not? 
Our WiFi receive buffer is enough to accept 802.11 packet, do you mean this buffer?
For TLS handshaking being slow, one possible reason is that our text code are put into SDRAM in order to make space.

JanneKiiskila, 2017-07-04:
Hi  @Archcady 

I have placed the white antenna to the board and the phone I'm using to host the Wifi hotspot for it is within centimetres of the Ameba board.

Archcady, 2017-07-04:
Hi @JanneKiiskila , 
From my side I didn't meet poor WiFi connection problem. Then, this is a issue I should research at.

JanneKiiskila, 2017-07-04:
Hi @Archcady 

I found out that if I just repeatedly restart the `mbed-os-example-client`, I would get various errors in the initial Wifi-stage. If you do 10 reboots (and I doubt one needs the `mbed-os-example-client` for this, as it's happening so early). 

There also seem to be some random crashes/freezes as well, we should found out the root cause for those. I.e. if you run the client 10 times, it should always end up with the same end result. Now it does not.

JanneKiiskila, 2017-07-04:
With UDP  & logs disabled I seem to have better luck, multiple passes now.

```
=========================================================
Check boot type form eFuse
SPI Initial
Image1 length: 0x2d64, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1

Initializing WIFI ...
WIFI initialized
K
K
K
K

KIFI is already running

Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (REALTEK)
[EasyConnect] Connecting to WiFi jannek.iki.fi

RTL8195A[Driver]: set ssid [jannek.iki.fi] 

RTL8195A[Driver]: start auth to ec:1d:7f:92:6d:99

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=8)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address f0:03:8c:e7:a9:cd
[EasyConnect] IP address 192.168.1.149

SOCKET_MODE : UDP
Connecting to coap://api.connector.mbed.com:5684

Registered object successfully!

```

marcuschangarm, 2017-07-04:
@Archcady it might be easier to debug using the smaller example applicetion: https://github.com/ARMmbed/mbed-os/issues/4700

I believe these issues might be related.

JanneKiiskila, 2017-07-04:
Yes, indeed. The mbed-os-example-client is a bit complicated with all the LWM2M stuff and the problem is clearly the TLS handshake - that we can see from the logs. So, if you get #4700 resolved - it will likely resolve the mbed-os-client-example issue as well.

Good work @marcuschangarm  - smaller test case typically makes life easier.

Enabling logs with UDP also makes the success rate drop, so extra load from serial tracing seems to impact this as well.

marcuschangarm, 2017-07-11:
I've closed the other issue since it turned out to be unrelated. 

When I run the mbed-os-example-client example, it often crashes before the TCP socket is even created, so the fix in the other thread wont help.

@Archcady what tools do you use for debugging the RTL8195AM? We currently don't have pyOCD support for that board so we can't debug using gdb.

Archcady, 2017-07-11:
@marcuschangarm, sadly we don't have some decent debugging methods, making debugging progress really slow.

JanneKiiskila, 2017-07-11:
@0xc0170 @c1728p9  - could we get the PyOCD support? Everyone would be needing it...

JanneKiiskila, 2017-07-11:
@sg-  can we label is OSCA, due to the PyOCD dependency.
a

JanneKiiskila, 2017-07-17:
@Archcady - has now done fix proposal via, https://github.com/ARMmbed/mbed-os/pull/4754

Archcady, 2017-07-17:
@JanneKiiskila with #4754, TLS handshake should now work fine.

prashantrar, 2018-02-09:
Is there any work needed still on this issue from realtek side?

JanneKiiskila, 2018-02-09:
I would close this one for now and file new issues via new issues. No point in hanging this for much longer unless more details can be surfaced. Though I have to admit that least I haven't been able to do much stress testing with this board anyway and it would likely be beneficial to run the stress tests for example for WiFi.

closed 2018-02-09 by JanneKiiskila