ARMmbed/mbed-os #4692: [API] Request API change configure ADC sampling time

LMESTM, 2017-07-03:
### Description
There have been 2 recent fixes where we needed to increase  sampling time of ADCs #4326 and #4641, 

This raises the question of optionally configuring this sampling time from the application, for instance 
as MIN / MAX / DEFAULT or MEDIUM. That would require a modification of the AnalogIn.h API, whether in a new constructor variant or by adding a new method.

LMESTM, 2017-07-03:
Adding more comments and suggestions from @ohagendorf regarding ADC:

Maybe two features or three would be nice to have:

setting sampling time
setting resolution
But curently for a fast conversion there is another huge problem. We analysed the adc read function:

    float read() {
        lock();
        float ret = analogin_read(&_adc);
        unlock();
        return ret;
    }
and lock()+unlock() need much more time than analogin_read(). I had to look into the project documentation for the exact numbers but it was something around 1.5µs for the conversion and 5...6µs for calling lock()+unlock() (Nucleo_F446RE).
One idea would be to add another read function with a callback under timer control or use continuous mode. In this case a fast equidistant conversion corresponding to the maximum conversion rate of the hardware would be possible. Or in combination with the above function to set the sampling time with a slower timing.

farrenv, 2019-02-13:
I had a similar issue working with the Nordic nrf52840 dev kit.  I had to step into the analogin function until i reached analogin_api.c.  From here i had to modify the following from 10 microseconds to 40.  This immediately improved the results with a battery monitoring circuit that we are using.  It would be nice to not have to modify the OS to get this functionality.
[image]

cmonr, 2019-02-13:
@elm3 ?

elm3, 2019-02-13:
The request to extend the ADC interface is in the backlog of the appropriate team.  I have just pinged them asking if there are plans that I can share.

closed 2019-10-31 by ciarmcom