ARMmbed/mbed-os #4698: I encounter a problem while using HSI on stm32l151cb

suyiii, 2017-07-04:
Hello!

I want to use mbed in the stm32l151cb.I downloaded mbed for TARGET_NUCLEO_L152RE, and changed startup file and head files for stm32l151.

And then I want to use only HSI ,so I changed USE_PLL_HSE_EXTC and USE_PLL_HSE_XTAL to 0 in system_stm32l1xx.c.

After downloading the hex file into MCU, the program get into dead loop when it comes to wait_ms function in the process of initializing CPP class. I clicked the stop button on keil, the breakpoint seemed to always stop on the SysTick_Handler function on startup file.

How can I solve the problem？

Thank you for your help!

0xc0170, 2017-07-04:
cc @bcostm @adustm @LMESTM @jeromecoutant

suyiii, 2017-07-04:
I ignored the comment in system_stm32l1xx.c to uncomment VECT_TAB_SRAM because I need to relocate my vector Table in Internal SRAM.

It will not get into deep loop in startup file, but it get into deep loop in wait_ms function. 

It seems that I have not confirgured the HSI correctly.

Is it because I forget something?

suyiii, 2017-07-04:
Well, I known what happened.

In Hal_tick.h, it uses TIM5 as interrupt, but TIM5 doesn't exists in my MCU.

Thank you for your attention!

I will close the issue.

LMESTM, 2017-07-04:
@suyiii thanks for sharing the issue you faced and the actual root cause of the problem !

suyiii, 2017-07-05:
Sorry, I am afraid that I have not solve the problem correctly.

I used TIM10 in hal_tick.h. It seems that the function us_ticker_read() always returns 0. Maybe TIM_MST->CNT never changes?

LMESTM, 2017-07-05:
That's ok. 
But it may be hard to guess where your issues come ... possible cause for CNT not increasing are: 
- timer not clocked or under reset (either up in the clock tree or the timer itself)
     => Could you share the modified hal_tick.h file ? 
- timer not properly initialized
    => can you check return codes of calls to HAL_TIM_OC_Start in hal_tick_23b.c file ?

bcostm, 2017-07-05:
I see that TIM10 is a 16-bit timer and not 32-bit. This timer has never been used as a 16-bit timer. Only timers 1, 4 and 21. So maybe there is something missing for this timer in the configuration ? (just a thought...)

suyiii, 2017-07-10:
Thank you for your help! I've left some days for some personal reason.
@LMESTM 
Below are my hal_tick.h codes.

#define TIM_MST      TIM10
#define TIM_MST_IRQ  TIM10_IRQn
#define TIM_MST_RCC  __TIM10_CLK_ENABLE()

#define TIM_MST_RESET_ON   __TIM10_FORCE_RESET()
#define TIM_MST_RESET_OFF  __TIM10_RELEASE_RESET()

#define TIM_MST_16BIT  1 // 1=16-bit timer, 0=32-bit timer

#define TIM_MST_PCLK  2 // Select the peripheral clock number (1 or 2)

#define HAL_TICK_DELAY (1000) // 1 ms

I will check return codes of calls to HAL_TIM_OC_Start later.
@bcostm 
Yes, TIM10 is a 16-bit timer,but I am afraid that I can't find 32-bit timer in stm32l151cb.  TIM4 in stm32l151cb is 16-bit, too.

suyiii, 2017-07-10:
Sorry, I check the reference manual. TIM10 has only one channel, so I changed it to TIM9. But the us_ticker_read still returns 0.

suyiii, 2017-07-10:
Well.  I made a foolish mistake that I ignored the pre-compiled symbol so I used the compiled us_ticker_32b.o and hal_ticker_16b.o downloaded from official website. 

Anyway, thank you for your attention and your help!

closed 2017-07-10 by suyiii