ARMmbed/mbed-os #4700: Realtek RTL8195AM unable to pass the TLS client example

marcuschangarm, 2017-07-04:
### Description
- Type: Bug 
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
Realtek RTL8195AM

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4.1 20160609

**mbed-cli version:** 
1.1.1

**meed-os sha:**
https://github.com/Archcady/mbed-os/commit/a3a7846817d7eefe59731762f681f53d2fc80da7

**DAPLink version:**
0241

**Expected behavior**
The Realtek RTL8195AM should be able to connect using TLS.

**Actual behavior**
Device hangs during hand shake:

```
=========================================================
Check boot type form eFuse
SPI Initial
Image1 length: 0x2d64, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1

Initializing WIFI ...
WIFI initialized

WIFI is already running
RTL8195A[Driver]: set ssid [mbed] 

RTL8195A[Driver]: start auth to 18:d6:c7:52:5f:1e

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=8)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
Using Ethernet LWIP
Client IP Address is 192.168.0.108
Connecting with developer.mbed.org
Starting the TLS handshake...

RTL8195A[Driver]: no beacon for a long time, disconnect or roaming
```

**Steps to reproduce**
Use modified mbed-os-example-tls tls-client:
https://github.com/marcuschangarm/mbed-os-example-tls/tree/master/tls-client

Set wifi credentials in mbed_app.json.

marcuschangarm, 2017-07-04:
@0xc0170 @Archcady @JanneKiiskila

JanneKiiskila, 2017-07-04:
Well, it seems to be so that it fails with TCP - would be nice for the test to indicate which transport it is using. mbed-os-example-client I have managed to get to register (which implies it succesfully did a TLS then), but only over UDP.  But yes, this would be a blocking bug for the mbed-os-example-client and mbed Cloud Client to work.

marcuschangarm, 2017-07-04:
@JanneKiiskila  but you couldn't get it to reliably connect over UDP, right? I think as a first step it would be good to get this example application to connect consistently.

JanneKiiskila, 2017-07-04:
Over UDP without logs it worked multiple times, over TCP - no success.

JanneKiiskila, 2017-07-04:
But, I agree - getting this TLS app to work is for sure a good starting point. Typically it has been one of the entry criteria for the board wake-ups.

marcuschangarm, 2017-07-04:
It sounds more and more like a memory problem to me...

bulislaw, 2017-07-05:
Probably worth adding test for that...

sg-, 2017-07-07:
We have tests for DTLS handshake, not sure what is different for the TCP version but could easily be extended.

@marcuschangarm Why are you on Archcady's fork and not a specific mbed OS release? Are there things needed in his repo?

marcuschangarm, 2017-07-09:
@sg- it contains a RAM fix that hasn't been merged yet: https://github.com/ARMmbed/mbed-os/pull/4665

JanneKiiskila, 2017-07-10:
Yep, you can't link it unless that fix is in.

JanneKiiskila, 2017-07-10:
@bulislaw  - This PR will get you WiFi to TLS as well; https://github.com/ARMmbed/mbed-os-example-tls/pull/101 and/or any other bearer we can figure out, as long as we just add them to easy-connect.

Archcady, 2017-07-10:
During my debugging, just find that if not set blocking, (_timeout == -1), then the example will have success output
```
<RTL8195A>
=========================================================

ROM Version: 0.2

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003) 

=========================================================

Check boot type form eFuse

SPI Initial

Image1 length: 0x2d64, Image Addr: 0x10000bc8

Image1 Validate OK, Going jump to Image1

Initializing WIFI ...

WIFI initialized

WIFI is already running

RTL8195A[Driver]: set ssid [RealEZ] 

RTL8195A[Driver]: start auth to 9c:1c:12:13:cd:01

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=3)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
Using Ethernet LWIP
Client IP Address is 172.25.24.144
mbedtls_ssl_set_hostname: ret = 0
Connecting with developer.mbed.org
Starting the TLS handshake...
TLS connection to developer.mbed.org established
Server certificate:
    cert. version     : 3
    serial number     : 65:7B:6D:8D:15:A5:B6:86:87:6B:5E:BC
    issuer name       : C=BE, O=GlobalSign nv-sa, CN=GlobalSign Organization Validation CA - SHA256 - G2
    subject name      : C=GB, ST=Cambridgeshire, L=Cambridge, O=ARM Ltd, CN=*.mbed.com
    issued  on        : 2017-04-03 13:54:02
    expires on        : 2018-05-06 10:31:02
    signed using      : RSA with SHA-256
    RSA key size      : 2048 bits
    basic constraints : CA=false
    subject alt name  : *.mbed.com, mbed.org, *.mbed.org, mbed.com
    key usage         : Digital Signature, Key Encipherment
    ext key usage     : TLS Web Server Authentication, TLS Web Client Authentication
Certificate verification passed

HTTPS: Received 440 chars from server
HTTPS: Received 200 OK status ... [OK]
HTTPS: Received 'Hello world!' status ... [OK]
HTTPS: Received message:

HTTP/1.1 200 OK
Server: nginx/1.11.10
Date: Mon, 10 Jul 2017 07:56:56 GMT
Content-Type: text/plain
Content-Length: 14
Connection: keep-alive
Last-Modified: Fri, 27 Jul 2012 13:30:34 GMT
Accept-Ranges: bytes
Cache-Control: max-age=36000
Expires: Mon, 10 Jul 2017 17:56:56 GMT
X-Upstream-L3: 172.17.0.5:80
X-Upstream-L2: developer-sjc-indigo-2-nginx
Strict-Transport-Security: max-age=31536000; includeSubdomains

Hello world!
```

JanneKiiskila, 2017-07-10:
@Archcady  - where exactly do you do that `_timeout` modification?

Archcady, 2017-07-10:
Hi @JanneKiiskila , I comment out ` _tcpsocket->set_blocking(false);` in HelloHTTPS(), located in main.cpp

marcuschangarm, 2017-07-11:
Thank you @Archcady, I can confirm that this change allows the test to complete.

I was hoping this would bring us closer to solving https://github.com/ARMmbed/mbed-os/issues/4683 but it looks like they are unrelated after all. 

I'm closing this issue and we can continue on the other thread.

closed 2017-07-11 by marcuschangarm

JanneKiiskila, 2017-07-11:
Well, why would this be a good solution? I don't think this case can be closed.

When setting the socket blocking status to false, we're activating the async API, right? Async API must work for the Client to work. Or am I misunderstanding something completely here? @kjbracey-arm

marcuschangarm, 2017-07-11:
It depends on what hat you are wearing. 😄 

For the mbed client, it is a blocker that we don't have async sockets and that the RTL8195AM crashes randomly. 

We need to be tracking both of these issues but I think repurposing this thread would be misleading.

JanneKiiskila, 2017-07-11:
Well, for the scope & single test case for this being the TLS test -> which it fails now -> this error is still valid. If it passes by modifying the test case -> it's not the same thing anymore.

But, yes, for the crashes -> we should file a new issue. But, we would really need the stack backtrace/PC etc. to be of any use for debugging. Which you can't get, until gdb works. So, bit of a chicken and egg.

ChiefBureaucraticOfficer, 2017-08-22:
@marcuschangarm - Can you validate now?

marcuschangarm, 2017-08-22:
With this branch: https://github.com/Archcady/mbed-os/commit/1e398cfc60399b36b002aacc360551a84b94cbfd

I'm getting this error:

```
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x3003D34C, task name: (null))

[mbed_die]  0x0 die here
```

v2422, 2017-08-25:
Hi @marcuschangarm,

Could you help to describe more about your environment?
I tried to download ARMCC6 for Linux from
https://developer.arm.com/products/software-development-tools/compilers/arm-compiler/downloads/version-6

There is no executable file "armcc" inside but "armclang". I tried to build the mbed tree but got some errors,
1. mbed use armcc not armclang
2. After I added a soft-link armcc to armclang, the compiling argument is not accepted.
Do I miss anything?