ARMmbed/mbed-os #4702: SD-Driver in mbed OS is having performance issue

yogpan01, 2017-07-05:
### Description
- Type: Bug 
SD Driver performance is not the same as in freeRTOS. There has been performance improvements done through #4207 however, accompanied changes that are probably required on the SD-Driver have not implemented/merged/tested yet.
Based on @sg- 's comments from discussion
"The last commit was mid-May. It has not yet been updated to use multi-byte write. It’ll be a few more days until someone gets to this. In the meantime, the simplest change to see improvement would be to replace https://github.com/ARMmbed/sd-driver/blob/master/features/filesystem/sd/SDBlockDevice.cpp#L579-L581 with the block write. I’m sure there are other areas that would benefit from this but these are the 2 major areas. And the read section https://github.com/ARMmbed/sd-driver/blob/master/features/filesystem/sd/SDBlockDevice.cpp#L559-L561"
- Related issue: `IOTMORF-991`

- Priority: Critical

yogpan01, 2017-07-05:
@sg-  @galc FYI

sg-, 2017-07-05:
Fixed by https://github.com/ARMmbed/sd-driver/pull/6, https://github.com/ARMmbed/sd-driver/pull/9, https://github.com/ARMmbed/sd-driver/pull/17, https://github.com/ARMmbed/sd-driver/pull/18, https://github.com/ARMmbed/sd-driver/pull/26, https://github.com/ARMmbed/sd-driver/pull/27

0xc0170, 2017-07-13:
> Fixed by ARMmbed/sd-driver#6, ARMmbed/sd-driver#9, ARMmbed/sd-driver#17, ARMmbed/sd-driver#18, ARMmbed/sd-driver#26, ARMmbed/sd-driver#27

@yogpan01 Is there any other improvements you are waiting for or anything that we should be aware of?

closed 2017-08-16 by sg-