ARMmbed/mbed-os #4704: ESP8266 starts to fail after some use in Continuous Integration

yogpan01, 2017-07-05:
### Description
- Type: Bug
After a few days/couple of weeks of use in test application's CI builds, ESP8266 modules start to fail tests. This blocks wlan tests using this module in CI.
The failures don't seem to follow any exact pattern. Most often either interface `connect()` or socket `connect()` fails, but also for example interface `disconnect()` has been seen to fail.
Taking one of the failing devices from the CI setup and testing locally the same behavior happens. It seems that often the first few connect attempts after reset fail and after that it mostly works – in CI this means quite consistent failures because the connect is only done once and once the test fails the device is reset.

- Related issue: `IOTMORF-976`
- Priority: Critical

geky, 2017-07-06:
Related pr on esp8266:
https://github.com/ARMmbed/esp8266-driver/issues/33

The tests need retry logic added.
cc @deepikabhavnani, @kegilbert

deepikabhavnani, 2017-07-07:
@yogpan01 - Please let us know the version of esp8266 driver and mbed-os are failing.

I am testing connect in loop and it is working fine with below versions.
mbed-os - mbed-os-5.5.2 tag
esp8266 – v1.3 tag

Also, firmware upgrade for esp8266 device is required:  https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update

0xc0170, 2017-07-24:
@yogpan01 Update for the above answers?

yogpan01, 2017-08-15:
@tommikas this fix is in reply to the issue that you raised here IOTMORF-976. Can you please verify and provide your input ?

tommikas, 2017-08-15:
@yogpan01, I implemented the retry logic suggested by @geky and the tests have been running in CI for the past 8 days without issue. I'll continue to monitor the situation but consider this resolved for now.

yogpan01, 2017-08-15:
Thanks @tommikas , I will close this issue for now.

closed 2017-08-15 by yogpan01