ARMmbed/mbed-os #4705: [M487] Mis-judge as error in mbed-os-tests-mbed_hal-flash

ccli8, 2017-07-05:
### Description
- Type: Bug
- Related issue: https://github.com/ARMmbed/mbed-os/pull/4608#pullrequestreview-47827294
- Priority: Minor

---------------------------------------------------------------
## Bug
The test case `mbed-os-tests-mbed_hal-flash`/`Flash - clock and cache test` is to detect illegal change of cache setting in flash algorithm. It sets up 0.1% performance drift rate to judge illegal change or not. But on NUMAKER_PFM_M487 target, the performance drift rate could reach 5%.
```
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash                                                         | Flash - buffer alignment test                                 | 1      | 0      | OK     | 0.12               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash                                                         | Flash - clock and cache test                                  | 0      | 1      | FAIL   | 0.18               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash                                                         | Flash - erase sector                                          | 1      | 0      | OK     | 0.14               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash                                                         | Flash - init                                                  | 1      | 0      | OK     | 0.08               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash                                                         | Flash - mapping alignment                                     | 1      | 0      | OK     | 0.05               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash                                                         | Flash - program page                                          | 1      | 0      | OK     | 0.21               |
```

**Target**
NUMAKER_PFM_M487

**Toolchain:**
ARM|uARM

**meed-os sha:**
mbed-os-5.5.0

theotherjimmy, 2017-07-05:
@ccli8 Why would the performance drift rate reach 5%?

ccli8, 2017-07-10:
> Why would the performance drift rate reach 5%?

@theotherjimmy It depends on flash controller in the bottom layer.

0xc0170, 2017-07-10:
Can you elaborate @ccli8 ? We have tested already few variants of flash controllers, haven't experienced this drift. What is causing it? Is this documented anywhere?

cc @c1728p9

chrissnow, 2017-07-13:
This may be resolved as #4666 is now merged.

ccli8, 2017-07-14:
@chrissnow Thanks for the fix.

@0xc0170 I check the fix on **NUMAKER_PFM_M487** target and the failed test passes now. So the cause is inline, not flash controller.
### ARM toolchain
```
+----------------------+------------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| target               | platform_name    | test suite                   | test case                     | passed | failed | result | elapsed_time (sec) |
+----------------------+------------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - buffer alignment test | 1      | 0      | OK     | 0.14               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - clock and cache test  | 1      | 0      | OK     | 0.15               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - erase sector          | 1      | 0      | OK     | 0.11               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - init                  | 1      | 0      | OK     | 0.13               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - mapping alignment     | 1      | 0      | OK     | 0.04               |
| NUMAKER_PFM_M487-ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - program page          | 1      | 0      | OK     | 0.21               |
+----------------------+------------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
```

### uARM toolchain
```
+-----------------------+------------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| target                | platform_name    | test suite                   | test case                     | passed | failed | result | elapsed_time (sec) |
+-----------------------+------------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
| NUMAKER_PFM_M487-uARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - buffer alignment test | 1      | 0      | OK     | 0.14               |
| NUMAKER_PFM_M487-uARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - clock and cache test  | 1      | 0      | OK     | 0.14               |
| NUMAKER_PFM_M487-uARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - erase sector          | 1      | 0      | OK     | 0.13               |
| NUMAKER_PFM_M487-uARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - init                  | 1      | 0      | OK     | 0.14               |
| NUMAKER_PFM_M487-uARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - mapping alignment     | 1      | 0      | OK     | 0.05               |
| NUMAKER_PFM_M487-uARM | NUMAKER_PFM_M487 | mbed-os-tests-mbed_hal-flash | Flash - program page          | 1      | 0      | OK     | 0.22               |
+-----------------------+------------------+------------------------------+-------------------------------+--------+--------+--------+--------------------+
```

closed 2017-07-14 by ccli8