ARMmbed/mbed-os #4712: Operator new[] out of memory causes Greentea test failure

akrnvm, 2017-07-06:
"features-tests-filesystem-mbr_block_device" test case is failing in GCC_ARM toolchain.
### Description
- Type: Question
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom board with limited RAM size.
RAM=16KB
ROM=128KB

**Toolchain:**
GCC_ARM

**Toolchain version:**
Version 6.0
**mbed-cli version:** 
(1.1.1)

**Expected behavior**
'Testing mbr read write' test case should pass

**Actual behavior**
 'Testing mbr read write' is fail due to 'Operator new[] out of memory'

**Steps to reproduce**

  mbed test -m XXXX -t GCC_ARM -v -n features-tests-filesystem-mbr_block_device

----------------------------------------------------------------
Can you please explain how to fix this issue in my custom board?
"features-tests-filesystem-mbr_block_device" and "tests-mbedmicro-rtos-mbed-threads" test cases are failing in greentea test with GCC_ARM toolchain.
In startup file, STACK = 0x400 and HEAP=0 are configured.

Observations:
Test case is passing with ARM and IAR compilers.
Test case is passing when configuring  "defaultlib = small" in target.json file

```
[1499314585.12][CONN][RXD] >>> Running 3 test cases...
[1499314585.18][CONN][INF] found KV pair in stream: {{__testcase_name;Testing formatting of master boot record}}, queued...
[1499314585.23][CONN][INF] found KV pair in stream: {{__testcase_name;Testing mbr attributes}}, queued...
[1499314585.27][CONN][RXD]
[1499314585.27][CONN][INF] found KV pair in stream: {{__testcase_name;Testing mbr read write}}, queued...
[1499314585.34][CONN][RXD] >>> Running case #1: 'Testing formatting of master boot record'...
[1499314585.41][CONN][INF] found KV pair in stream: {{__testcase_start;Testing formatting of master boot record}}, queued...
[1499314585.48][CONN][INF] found KV pair in stream: {{__testcase_finish;Testing formatting of master boot record;1;0}}, queued...
[1499314585.55][CONN][RXD] >>> 'Testing formatting of master boot record': 1 passed, 0 failed
[1499314585.56][CONN][RXD]
[1499314585.61][CONN][RXD] >>> Running case #2: 'Testing mbr attributes'...
[1499314585.65][CONN][INF] found KV pair in stream: {{__testcase_start;Testing mbr attributes}}, queued...
[1499314585.69][CONN][RXD] partition 1 partition number: 1
[1499314585.72][CONN][RXD] partition 1 partition start: 0x200
[1499314585.76][CONN][RXD] partition 1 partition stop: 0x1000
[1499314585.79][CONN][RXD] partition 1 partition type: 0x83
[1499314585.83][CONN][RXD] partition 1 read size: 512 bytes
[1499314585.86][CONN][RXD] partition 1 program size: 512 bytes
[1499314585.90][CONN][RXD] partition 1 erase size: 512 bytes
[1499314585.93][CONN][RXD] partition 1 size: 3584 bytes
[1499314585.96][CONN][RXD] partition 2 partition number: 2
[1499314586.00][CONN][RXD] partition 2 partition start: 0x1000
[1499314586.04][CONN][RXD] partition 2 partition stop: 0x2000
[1499314586.07][CONN][RXD] partition 2 partition type: 0x83
[1499314586.11][CONN][RXD] partition 2 read size: 512 bytes
[1499314586.14][CONN][RXD] partition 2 program size: 512 bytes
[1499314586.18][CONN][RXD] partition 2 erase size: 512 bytes
[1499314586.21][CONN][RXD] partition 2 size: 4096 bytes
[1499314586.27][CONN][INF] found KV pair in stream: {{__testcase_finish;Testing mbr attributes;1;0}}, queued...
[1499314586.32][CONN][RXD] >>> 'Testing mbr attributes': 1 passed, 0 failed
[1499314586.32][CONN][RXD]
[1499314586.37][CONN][RXD] >>> Running case #3: 'Testing mbr read write'...
[1499314586.42][CONN][INF] found KV pair in stream: {{__testcase_start;Testing mbr read write}}, queued...
[1499314586.45][CONN][RXD] Operator new[] out of memory
[1499314595.60][HTST][INF] test suite run finished after 10.54 sec...
[1499314595.60][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1499314595.61][HTST][INF] CONN exited with code: 0
[1499314595.61][HTST][INF] No events in queue
[1499314595.61][HTST][INF] stopped consuming events
[1499314595.61][HTST][INF] host test result(): None
[1499314595.61][HTST][WRN] missing __exit event from DUT
[1499314595.61][HTST][WRN] missing __exit_event_queue event from host test
[1499314595.61][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1499314595.61][HTST][INF] calling blocking teardown()
[1499314595.61][HTST][INF] teardown() finished
[1499314595.61][HTST][INF] {{result;timeout}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test suite 'features-tests-filesystem-mbr_block_device' ...................................... TIMEOUT in 48.81 sec
        test case: 'Testing formatting of master boot record' ........................................ OK in 0.07 sec
        test case: 'Testing mbr attributes' .......................................................... OK in 0.62 sec
        test case: 'Testing mbr read write' .......................................................... ERROR in 0.00 sec
mbedgt: all tests finished!
```

-----------------------------------------------------------------

## Question

Can you please explain how to resolve the HEAP memory issue for these test cases.

bulislaw, 2017-07-06:
`Operator new[] out of memory` seems like you need more memory on the HEAP. GCC usually generates bigger code than other toolchains, so I'm guessing that's why you are having the problems. You may need to find some savings on your platform or possibly try to optimize the test itself. One thing to note is that for GCC (and ARM) HEAP size is dynamic, so any savings elsewhere should be automatically available for the HEAP to grow. You may want to rethink main thread stack size, as by default it's 4K which is excessive for tests (especially for small platforms), but we use it as default for all platforms so they feel the same to the user. You can try to make the main stack smaller for this test only https://github.com/ARMmbed/mbed-os/blob/master/rtos/mbed_boot.c#L182

pan-, 2017-07-10:
@bulislaw duplicate of #4591 .

akrnvm, 2017-07-11:
@bulislaw Thank you for the comment.
Issue got resolved.

akrnvm, 2017-08-14:
hi ,
I have modified the size of  main thread stack size to 2K and the testcases are PASS.
But I saw some other controllers with same size(RAM 16KB) is executing without any changes.
After build : 
```
+------------+-------+------+-----------+-------------+
| static_ram | stack | heap | total_ram | total_flash |
+------------+-------+------+-----------+-------------+
|       9632 |  1024 |    0 |     10656 |       67655 |
+------------+-------+------+-----------+-------------+
```
Is this RAM size is enough to execute this testcase?
how can i resolve this issue.
Is any generic solution is available without modifying application.

closed 2017-08-14 by akrnvm