ARMmbed/mbed-os #4713: The error message for mbed OS 5.5 disabling Cortex-A is different between online compiler and offline (mbed CLI)

TomoYamanaka, 2017-07-06:
### Description
- Type: Bug
- Related PR: https://github.com/ARMmbed/mbed-os/pull/4415, https://github.com/ARMmbed/mbed-os/pull/4377
- Priority: Major

**Target**
RZ_A1H | VK_RZ_A1H

---------------------------------------------------------------
## Bug

mbed OS 5.5 disable building of Cortex-A targets.
Therefore, As shown in the https://github.com/ARMmbed/mbed-os/pull/4415 and https://github.com/ARMmbed/mbed-os/pull/4377,
we Renesas discussed to ARM mbed team, and decided to appear the below error message when users built mbed OS 5.5.
```
Could not compile for RZ_A1H: RZ_A1H Will be supported in mbed OS 5.6. 
To use the RZ_A1H, please checkout the mbed OS 5.4 release branch. 
See https://developer.mbed.org/platforms/Renesas-GR-PEACH/#important-notice for more information
```

After mbed OS 5.5 was released,
In offline(mbed CLI), we comfirmed the above message appeared to terminal.
This offline message is match what we expected.
However, in Online compiler, the below error message apeared.
` Cannot open source input file "cmsis_os2.h": No such file or directory "#include "cmsis_os2.h""`
This online message is **not** match hat we expected.

Could you give me your opinion about why the error message is defferent?
We expect that the same message appears both online compiler and offline (mbed CLI).

bulislaw, 2017-07-06:
@theotherjimmy it seems something slips past the checks? Which it shouldn't as the mbed OS 5 is removed from the supported OS versions...

theotherjimmy, 2017-07-06:
Hi @TomoYamanaka I'm currently working on updating the version of the tools on the website. @bulislaw It's not that it slipped through, it's that the web tools are just behind.

theotherjimmy, 2017-07-06:
Further status: My development version shows the error message like expected.

TomoYamanaka, 2017-07-07:
@theotherjimmy

Thank you for your comments.

If you are possible, please can you let me know an approximate time that version of the tools on the website will be updated?

theotherjimmy, 2017-07-07:
I can tell you that they will be up to 5.5.1. @immunda Will be able to tell you when it's live.

toyowata, 2017-07-19:
Now, I see correct error message in online IDE.

@masaohamanaka 
Please check it again and close this ticket once you confirm the fix.

TomoYamanaka, 2017-07-20:
I will close because I confirmed the fix of this problem.

closed 2017-07-20 by TomoYamanaka