ARMmbed/mbed-os #4716: Stream::write always returns at least 1

marogi, 2017-07-06:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
All

**Toolchain:**
Any (tested on GCC_ARM)

**Toolchain version:**
Any (tested on  6-2017-q1-update/ARM/embedded-6-branch revision 245512)

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
1af8aef5dc0dd981f23bfe3cfdfb9df8d6e820f9

**Expected behavior**
`Stream::write(const void* buffer, size_t length)` should return value as specified on [mbed::Stream Class Reference](https://docs.mbed.com/docs/mbed-os-api/en/mbed-os-5.5/api/classmbed_1_1Stream.html)
>""
>Returns
>The number of characters written (possibly 0) on success, -1 on error.
>""

**Actual behavior**
`Stream::write` will return 0 only if length is 0. Otherwise, independent of success or failure  of `_putc` call, `const char* ptr` is always incremented and return value (`return ptr - (const char*)buffer;`) will be at least 1 (even if `__putc` fails - returns `EOF` - for first character).

**Steps to reproduce**
Check return status of `Stream::putc(int c)` on uninitialized stream (for example disconnected USB or uninitialized UART).

----------------------------------------------------------------
## Current code:
```cpp
ssize_t Stream::write(const void* buffer, size_t length) {
    const char* ptr = (const char*)buffer;
    const char* end = ptr + length;

    lock();
    while (ptr != end) {
        if (_putc(*ptr++) == EOF) {
            break;
        }
    }
    unlock();

    return ptr - (const char*)buffer;
}
```

Proposed fix:
```cpp
ssize_t Stream::write(const void* buffer, size_t length) {
    const char* ptr = (const char*)buffer;
    const char* end = ptr + length;

    lock();
    while (ptr != end) {
        if (_putc(*ptr) == EOF) {
            break;
        }
        ptr++; // ptr is incremented only on __putc success
    }
    unlock();

    return ptr - (const char*)buffer;
}
```

0xc0170, 2017-07-12:
cc @kjbracey-arm

kjbracey-arm, 2017-07-12:
Stream has a wide variety of problems - it doesn't correctly implement FileHandle as it is now specified, and changing that would cause backwards compatibility issues. It (and hence Serial) isn't particularly useful for real interactive use (eg `read(X,N)` always tries to transfer N bytes rather than as much as is available).

We are planning to deprecate Stream and Serial, in favour of implementing FileHandle directly and using UARTSerial. (@0xc0170 - guess we should press on with that?)

This particular fix seems innocuous, and I'd be okay with it, but as I recall the general consensus during #4119 was to not change Stream's behaviour at all - if users want backwards-compatible behaviour, they can stick with Stream, but if they want working behaviour, move to better/correct FileHandle implementations such as UARTSerial.

marogi, 2017-07-12:
@kjbracey-arm : Thank you for clarification. 

I was not aware of Stream deprecation plan and I stumbled upon this issue on my fork, so I can have the fix there and move to FileHandle when switching back to official repo.

Issue can be closed.

kjbracey-arm, 2017-07-12:
Great. Stream doesn't really make it much easier to implement FileHandle - I decided that there was no particular merit in inventing a new similar "FileHandle to putc/getc" adapter.

And instead of the printf helpers, users can now use fdopen() to get a real FILE * from a FileHandle *, which they can then use for any <stdio.h> call.

Nodraak, 2017-07-12:
Closing my PR then

0xc0170, 2017-07-13:
Thank you all for the inputs for this.

closed 2017-07-13 by 0xc0170