ARMmbed/mbed-os #4723: Can't define custom_target

rene-aguirre, 2017-07-07:
"custom_targets" mbed_app.json configuration can't be set, mbed cli reports this error:

```
[ERROR] Unknown key(s) 'custom_targets' in mbed_app.json
```

It looks the file `mbed-os/tools/config/__init__.py` is missing the definition in the `__allowed_keys` for application items.

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom / Any

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
latest @Jul/6th/2017

**mbed-cli version:** 
1.2.0

**meed-os sha:**
8828635da (HEAD, tag: mbed_lib_rev146, tag: mbed-os-5.5.2, tag: latest, origin/mbed-os-5.5) Merge pull request #4689 from ARMmbed/release-candidate

**Steps to reproduce**
Add this to `mbed_app.json`
{
    "custom_targets": {
        "DISCO_QEMU": {
            "inherits": ["DISCO_F429ZI"],
            "extra_labels_add": ["DISCO_F429ZI"],
            "overrides":{"core": "Cortex-M4"}
        }
    }
}

Try to build any valid target (expecting this configuration to be passive).

theotherjimmy, 2017-07-07:
use custom_target.json as described here: https://github.com/ARMmbed/Handbook/blob/5.5/docs/advanced/mbed_targets.md

theotherjimmy, 2017-07-07:
@rene-aguirre Where did you find documentation that lead to trying this?

rene-aguirre, 2017-07-07:
Here:
https://docs.mbed.com/docs/mbedmicro-api/en/latest/api/md_docs_config_system.html

I googled for "mbed custom targets" and ended up in that page.

theotherjimmy, 2017-07-07:
Thanks. Adding a deprecated notice.

theotherjimmy, 2017-07-07:
To get this working, create a file named `custom_targets.json` in the root of you project with the following contents:
```json
{
    "DISCO_QEMU": {
        "inherits": ["DISCO_F429ZI"],
        "core": "Cortex-M4"
    }
}
```
Note: This definition omits the `"extra_labels_add"` bit as the labels of a target are inherited.

Alternatively, if you just need to override the core attribute and don't need it to be a custom target, you could have this in your `mbed_app.json`:
```json
{
    "target_overrides": {
        "DISCO_F429ZI": {
            "target.core": "Cortex-M4"
        }
    }
}
```

theotherjimmy, 2017-07-07:
I hope the above helps @rene-aguirre. I am now using this issue to track updating the documentation to avoid this issue in the future.

rene-aguirre, 2017-07-07:
thanks @theotherjimmy , inheriting from FAMILY_STM32 doesn't work as CMSIS include paths are missing, but the custom_targets.json with the DISCO_F429ZI override works as expected.

theotherjimmy, 2017-07-07:
Excellent. As it turns out, the page you found is available by mistake. It should be taken down shortly. Is there any documentation that you would like to see added to https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/config_system/ that would make it more clear that defining custom targets is now in its own file?

rene-aguirre, 2017-07-07:
Definitely! :-) That page is missing any reference to `custom_targets.json` it only mentions `mbed_app.json` and `mbed_lib.json`.

When I looked at that page I had initially thought the only way to have a custom targets was though editing the `targets.json` file due lack of any `custom_targets.json` reference.

theotherjimmy, 2017-07-07:
I'll add that soon, and mention you on the PR for review.

theotherjimmy, 2017-07-07:
@rene-aguirre see https://github.com/ARMmbed/Handbook/pull/19

theotherjimmy, 2017-07-07:
@rene-aguirre I think we got this all tidied up. I'm going to close this.

closed 2017-07-07 by theotherjimmy