ARMmbed/mbed-os #4724: mbed TLS examples failing with mbed-os-5.5.2

andresag01, 2017-07-07:
### Description
- Type: Bug
- Related issue: `mbed-os-5.5.2`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
Multiple targets tested by Jenkins CI

**Toolchain:**
GCC_ARM, ARM and IAR

**Toolchain version:**
Refer to [this](https://github.com/ARMmbed/mbed-os-example-tls/pull/99) CI job for more information regarding tools.

**Expected behavior**
htrun should be able to match the regular expression template to the output in the serial port for the tls-client example.

**Actual behavior**
The tls-client example prints the following error in the serial console:
```
mbedtls_ssl_read() failed: -0x7880 (-30848): SSL - The peer notified us that the connection is going to be close
```

0xc0170, 2017-07-10:
@andresag01 Has anyone looked at this, what is this regression? The cause?

andresag01, 2017-07-10:
@0xc0170: I have made @adbridge and @sg- aware of the problem. We are currently investigating the issue in mbed TLS, but I am not sure if anyone is looking at it from the mbed OS side.

andresag01, 2017-07-11:
We have identified a problem in the testing of our examples which was causing the failure. This is not directly related to mbed OS 5.5.2 so we have merged the update to mbed-os-example-tls. Closing this issue.

closed 2017-07-11 by andresag01