ARMmbed/mbed-os #4730: STM32L053xx DAC pin not correctly set in PeripheralPins.c

bcostm, 2017-07-10:
### Description
- Type: Bug
- Related issue: https://developer.mbed.org/questions/78531/DAC-error-Unknown-DAC-channel/
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
DISCO_L053C8
NUCLEO_L053R8
Check also all STM32 devices having one single DAC output

**Toolchain:**
ALL

**mbed-os sha:**
efc5c47e55b8a30c2acb366808975411e4bae7fd

----------------------------------------------------------------
**Suggested enhancement**

In PeripheralPins.c replace the line:
`{PA_4, DAC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 0, 0)}, // DAC_OUT`
with:
`{PA_4, DAC_1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 1, 0)}, // DAC_OUT`

theotherjimmy, 2017-07-10:
@bcostm Would you like to submit that pull request? Would you like me to?

LMESTM, 2017-07-10:
@bcostm  may not connect frequently in the coming days, 
so @theotherjimmy  do you think you could do it ?

theotherjimmy, 2017-07-10:
Yes. Just a moment.

theotherjimmy, 2017-07-10:
see PR #4732

PR #4732 (2017-07-17): Correct DAC pin assignment
### Description
- Type: Bug
- Related issue: https://developer.mbed.org/questions/78531/DAC-error-Unknown-DAC-channel/
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
DISCO_L053C8
NUCLEO_L053R8

**Toolchain:**
ALL

**mbed-os sha:**
efc5c47e55b8a30c2acb366808975411e4bae7fd

Resolves #4730 

@bcostm @LMESTM Could you review?
closed 2017-07-17 by theotherjimmy