ARMmbed/mbed-os #4735: NRF52: Adjust idle thread stack size

marisabele, 2017-07-11:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#4563`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM

**Expected behavior**
Expected to extend stack size to 512 for NRF52 targets but in reality effective only for TARGET_MCU_NRF51822
Issue introduced: https://github.com/ARMmbed/mbed-os/commit/970ee38079fc1c92da37ccc06e7cd1e8080446e5

0xc0170, 2017-07-11:
cc @pan- @nvlsianpu

pan-, 2017-07-11:
@marisabele It is only effective for `NRF52840` targets and it isn't effective for `TARGET_MCU_NRF51822`. 
Would it be necessary for `NRF52832` targets ?

pan-, 2017-07-11:
@marisabele I've submitted a fix, please review #4736

jrobeson, 2017-08-17:
did it fix the problem? the PR was merged a bit ago.

closed 2017-08-17 by 0xc0170