ARMmbed/mbed-os #4737: can not retrive

netanelgonen, 2017-07-11:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement 
- Related issue: #4526 
- Priority: Blocker

## Enhancement

cannot detect the file system handle that was  mount on a block device using the block device handle 

**Suggested enhancement**
add and api to get the file system handle

**Pros**
will be much easier to link file system to partition in case of multiple partitions

**Cons**
what if we are not mounting a file system on partition

netanelgonen, 2017-07-11:
@avolinski @amiraloosh @geky @galc @JanneKiiskila @liatvindzberg

JanneKiiskila, 2017-07-11:
@sg- - please label as OSCA!

JanneKiiskila, 2017-07-13:
IOTMORF-1018 internal ref.

geky, 2017-07-18:
Why do you want this? There's really no garuntee that a user has a filesystem object or is even using that space for a filesystem. The user may be nesting other partitioning schemes or using the partition for raw storage.

If you are just looking for a filesystem you know may exist, this is what the error on the filesystem mount function is for:
``` c++
FATFileSystem fs[2];
MBRBlockDevice part[2];
int count = 0;
for (int i = 0; i < 4; i++) {
    part[count] = MBRBlockDevice(i);
    int err = part.init(); // check for partition
    if (!err) {
        printf("partition %d exists\n", i);
        err = fs[i].mount(&part[count]); // check for FAT filesystem
        if (!err) {
            printf("FAT filesystem exists on partition %d\n", i);
            count++;
        }
    }
}
```

If you think this is ugly you're right. Trying to deduce what the user is up to with their storage is a scary idea and you're going to be much safer just asking them.

What if the user created their own FAT filesystem on a partition you weren't using and yours suddenly got corrupted. Would you just eat the user's filesystem?

Why not just use partitions that are explicitly specified by the user or implicitly specified by your API? From looking at your implementation for the FreeRTOS+ChanFS port, this is exactly what you're already doing:
https://github.com/ARMmbed/mbed-client-pal/blob/master/Source/Port/Reference-Impl/OS_Specific/FreeRTOS/FileSystem/pal_plat_fileSystem.c

netanelgonen, 2017-07-19:
Hi @geky ,
thank you for your response.

you are correct hit solution is quite ugly, but it was your advice so i continued with this option.
in freeRTOS and Linux we asking from the use one parameter only the base folder (path) to work on top of it. that's all.. this cannot be done trough the current implementation in mbedOS. 

i need this to use the mbedOS format API that is not forking on path like in freeRTOS but need to use the filesystem handle and the partition(MBR) handle and the path for mounting.

this can be done quite easily if we change the array in
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L134
to be a struct that contains the block device handle as well and then i can iterate trough the file systems look for the one with the same path and then to retrieve the block device handle as well.
is this looks like a better solution?

geky, 2017-07-20:
@netanelgonen and discussed this issue to try to clear things up. Here's my attempt to summarize the issue (@netanelgonen feel free to correct anything I get wrong).

1. The pal has to worry about two consumers for the filesystem: pal users (ie client) and mbed users (ie applications). These consumers may end up in the same system and should not corrupt each other's state. A good example of this an application that contains both its own filesystem and client. The user may not be aware of the filesystem in client, but would still be surprised if their own filesystem became corrupted.

2. Formatting a block device will destroy any filesystem that existed on the block device. This includes any filesystems the user created outside of the pal. If client formatted a partition that an mbed user had mounted a filesystem on, the mbed user's filesystem would then become corrupted.

3. mbed does not keep track of which filesystems are mounted on which block devices. This means it is currently impossible to unmount any filesystems that may be mounted on a given block device, without having direct access to the C++ object that describes that filesystem/mount point. This is different from the other pal ports: ChanFS contains an array of all disks, and Linux has a table of all mount points.

@netanelgonen's solution is to introduce a global table of mount points. This would match what's available in Linux and ChanFS and would fix the problem for pal.

I'm not currently sold on this fix as it still leaves the issue of non-filesystem consumers such as MBRBlockDevices or FTLs. Perhaps we would need a more fleshed out system for registering "resource" owners (what does Linux do here?). This also may have wider implications in regards to the other mbed APIs (should flash drivers register as SPI owners?) though that may be taking this idea too far for now.

The only other solution I can think of is to just not support formatting a block device that has a mounted filesystem.

netanelgonen, 2017-07-24:
Hi @geky any progress with this? how do we continue from here?

geky, 2017-07-25:
I'm still of the opinion that we shouldn't support this, at least not unless we implement a more generalized concept of resource owners. And even this I see only a little benefit.

The best use case I can think of is the one you pointed out: A user wants they want to "reset" a filesystem (like during a factory reset). I can only think of a few situations for this:
- A single user owns the filesystem - They can be held responsible for unmounting the filesystem before formatting.
- The filesystem is also owned by a bootloader - The bootloader will invalidate the user's filesystem anyways, so the bootloader should somehow signal the user's code that it needs to be remounted, probably by resetting the device. 
- A multi-user embedded system? I don't think mbed OS is there yet.

Here's a few other examples I could think of:
- A user may want to use pal format to mount a corrupted filesystem.

  In this case it's impossible to mount the filesystem (it's corrupted). There is no problem with calling pal format and then mounting the filesystem with mbed OS.

- A library may want to format a partition in the same MBR as a user's filesystem.

  This should be fine with the current API, as long as fixed partition numbers are used and the other partition entry is not deleted, the neighboring filesystem should be unaffected. I vehemently suggest not dynamically finding an unused partition for the above reasons.

Feel free to mention a use case I'm not thinking of. A concrete example where a user would need to format a mounted filesystem would probably convince me to change my mind. And sorry for the slow responses, this has been a interestingly complex issue and I've been stretched a bit thin.

netanelgonen, 2017-07-25:
@geky  @c1728p9 @avolinski  "at least not unless we implement a more generalized concept of resource owners" is this a possibility? 
i think that the simple case of holding qa mount table like a lot of other OS is pretty good use case for itself spicily this is almost zero effort.

as we discussed on the use case of pal multiple times by email ,by phone and in this thread. there is no option to perform format to a partition without declare globally the FileSystem and partition handles.
this is not reasonable

netanelgonen, 2017-08-09:
Hi After a talk with @sg- this is our ideal way performing the format:

fs = getFileSystemFromPath("PAL_ROOT_PATH");
bd = getBdFromFs(fs);
unmount(fs);
format(bd);
mount(fs);

the second best way is to use static:
fs = getFileSystemFromPath(PAL_ROOT_PATH);
bd = getBdFromNamePAL_DEVICE_NAME);
unmount(fs);
format(bd);
mount(fs);

geky, 2017-08-10:
Summary from offline discussion, we can do most of that, with the exception of choosing the format function at runtime. Will need to look into how that fits into C++.

---

> fs = getFileSystemFromPath("PAL_ROOT_PATH");

This can be achieved with the [FilePath class](https://github.com/ARMmbed/mbed-os/blob/0da63eef2b3debaed5db9776d1e913ae982ce7c3/platform/FilePath.h):
``` cpp
FilePath path("/fs");
if (!path.isFileSystem()) {
    error("not a filesystem");
}

FileSystem *fs = (FileSystem*)path.getFileSystem();
```

> bd = getBdFromFs(fs);

This will need to be added as an API to the FileSystem and all inheriting classes.

> unmount(fs);

``` cpp
fs.unmount();
```

> format(bd);
``` cpp
// Name of filesystem is needed at compile time
FATFileSystem::format(bd);
```

> mount(fs);
``` cpp
fs.mount(bd);
```

closed 2017-12-05 by netanelgonen