ARMmbed/mbed-os #4739: osSemaphoreAcquire hangs in LWIP with cellular

amq, 2017-07-11:
### Description
- Type: Bug
- Possibly related issues: #4575, #4537
- Priority: Major

---------------------------------------------------------------
## Bug

**Toolchain:**
Tested with GCC_ARM

**Toolchain version:**
6-2017-q1-update

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
8828635da469162cf2854b5287561c663fb96e72

Similar behavior also observed with c9e63f14085f5751ff5ead79a7c0382d50a813a2

**Expected behavior**
Can connect to the internet.

**Actual behavior**
The initial connection hangs with the following trace:

[image]

By pausing I get into the `Default_Handler` with this trace:

[image]

[This](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.5.2/features/FEATURE_LWIP/lwip-interface/lwip-sys/arch/lwip_sys_arch.c) is the relevant code:

```
u32_t sys_arch_sem_wait(sys_sem_t *sem, u32_t timeout) {
    u32_t start = us_ticker_read();
    
    if (osSemaphoreAcquire(sem->id, (timeout != 0)?(timeout):(osWaitForever)) != osOK)
        return SYS_ARCH_TIMEOUT;
    
    return (us_ticker_read() - start) / 1000;
}
```

Sometimes I can get past it by stopping with a breakpoint and continuing manually.

I could always successfully connect using 5.4.3 and 5.4.6 with a manually applied https://github.com/ARMmbed/mbed-os/pull/4119 pretty much at all stages of its development.

I'm using the configuration from https://github.com/ARMmbed/mbed-os-example-cellular

0xc0170, 2017-07-11:
cc @hasnainvirk @kjbracey-arm @c1728p9 @geky

kjbracey-arm, 2017-07-12:
Feels kind of like an OS bug, rather than anything that could be pinned on the lwip/cellular code. That is a timed wait. So how can it never unstick?

But on the other hand, OS bugs are pretty uncommon, just like compiler bugs.

So what could it be? Corruption of the semaphore while waiting? Failure to initialise it? Someone's broken the ticker clock?

Actually, can you confirm which one it's stopping on? Your initial trace seems to show it was stopping on the lwip_tcpip_inited wait, but then you say it was lwip_netif_linked.

amq, 2017-07-12:
@kjbracey-arm sorry, the first screenshot was taken at the wrong moment, I've corrected it now. It consistently fails with `lwip_netif_linked` and works with `lwip_tcpip_inited`.

kjbracey-arm, 2017-07-12:
Hmm, what's the second trace about? What do you mean "by pausing"? Are we looking at a crash during the wait, rather than it not coming out? Someone tried an sprintf and it died?

Possibly a stack overflow? We know the PPP stack is a bit small, and it can crash if there's a frame checksum error - see pull #4677 which addresses that.

amq, 2017-07-12:
When blinky (Ticker) stops, I know that the device hung, then I press pause in Eclipse.

Applying https://github.com/ARMmbed/mbed-os/pull/4677 seems to have helped a bit, I can always get past `sys_arch_sem_wait(&lwip_netif_linked, 15000)` now if I set a breakpoint there and continue manually, but without it, `mbed_lwip_bringup_2` still hangs.

kjbracey-arm, 2017-07-12:
That suggests some sort of timing-related issue. Try adding a couple-of-second software delay before the sem_wait, see if that changes anything.

Still not sure how it doesn't just continue after 15 seconds though, regardless of whatever else happens.

amq, 2017-07-14:
Tried `wait_ms(1)`, `wait_ms(5000)` and `wait_ms(10000)`, doesn't help. (I did add `#include "mbed_wait_api.h"`)

It is really strange that setting a breakpoint always works, even if I press continue immediately.

amq, 2018-01-28:
Can't replicate it at the moment, but I'm not sure the issue is completely gone. Closing until / if it can get replicated again.

closed 2018-01-28 by amq