ARMmbed/mbed-os #4743: SD SPI block read fails - K64F

deepikabhavnani, 2017-07-11:
### Description
- Type: Bug 
SPI block read fails for few SD cards, byte read was working fine.
Root caused it to incorrect dummy byte sent while performing SPI reads.
SD card requirement says 0xFF should be sent to perform read operations via SPI, in case of block read 0x00 is set as dummy data.

`#define DSPI_DUMMY_DATA (0x00U) /*!< Dummy data used for Tx if there is no txData. */`

- Priority: Blocker

**Target**
K64F

**Steps to reproduce**
Run SD card block read/write test with SD driver version : sd-driver-0.1.0-mbed-os-5.5.1

deepikabhavnani, 2017-07-11:
@geky @c1728p9

0xc0170, 2017-07-12:
I would use 0xFF as default value, also used by SPI async transfers, we could use that value - `SPI_FILL_WORD` by default. This would also be used if we extend API by being able to set this fill world, that was already discussed earlier that it would be nice to have.

cc @mmahadevan108

JanneKiiskila, 2017-07-12:
Please fix ASAP to an mbed OS 5.5.x patch release.

deepikabhavnani, 2017-07-12:
@0xc0170 - We can use same API "spi.dummy" as mentioned in PR #4744 for async transfers as well. At present changes are done just for "spi_master_block_write" HAL API, I would prefer changes in async transfer as new pull request.

JanneKiiskila, 2017-07-18:
Internnaly IOTMORF-1080

maclobdell, 2017-07-19:
@mmahadevan108

mmahadevan108, 2017-07-24:
@deepikabhavnani 
There is a new API available in the MCUXpresso SPI driver to set the dummy data.
void DSPI_SetDummyData(SPI_Type *base, uint8_t dummyData)

I will update the DSPI driver. Could you please update the K64F SPI driver to use this API as part of PR #4744

deepikabhavnani, 2017-07-24:
@mmahadevan108 - I believe DSPI_MasterTransferBlocking() API in block write will boost the performance right? Is it applicable to just K64F or other K series targets as well?

If change is required for all kinetics devices, I would prefer new PR. I will just need list of devices and support for all of them to have consistent behavior.

mmahadevan108, 2017-07-24:
I will create a separate PR to update the Kinetis SPI drivers to include an API to update the DUMMY_DATA variable.

However I am not clear where this API should be invoked from inside mbed.

deepikabhavnani, 2017-07-24:
@mmahadevan108 : After you add API to DSPI driver, I can update mbed-os code in same PR. If that is fine for you.

mmahadevan108, 2017-07-24:
PR for updating K64F DSPI driver is https://github.com/ARMmbed/mbed-os/pull/4805
I will work to update the other Kinetis platforms separately.

closed 2017-08-23 by deepikabhavnani