ARMmbed/mbed-os #4748: REALTEK_RTL8195AM - needs updated DAPLINK

JanneKiiskila, 2017-07-12:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#abc`
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**Toolchain:**
Irrelevant.

**Toolchain version:**
Irrelevant.

**mbed-cli version:** 

**mbed-os sha:**
Irrelevant.

**DAPLink version:**
https://developer.mbed.org/platforms/REALTEK-RTL8195AM/
DAP_FW_MBED_RTL8195AM_V0.bin
(hard to say which version)

IC ID:     Auto Reset:     1   0   Automation allowed:     Bootloader  Interface   Daplink Mode:    Version:   Bootloader Version:     Interface Version:  Git SHA:    e6046fae5e2d3433b04ed95bb80a96375e8db68b    Local Mods:     USB Interfaces:     MSD , CDC   , HID   Bootloader CRC: 0x  Interface CRC: 0x   øµ

It says in the binary.

**Expected behavior**

mbed-clitest reset command should work 100% reliably.
flashing should work 100% reliably.

**Actual behavior**

Reset does not work.
Board sometimes drops out of USB completely with Windows saying "USB device malfunctioned".

**Steps to reproduce**

Flash / reset device via DAPLINK functionality.

This blocks the usage of this board in the CI systems.
@jupe @ErkkiMehtala 
- please fill in details what API we use.

Likely fix would be to migrate to latest version of DAPLINK?

JanneKiiskila, 2017-07-12:
@Archcady

JanneKiiskila, 2017-07-12:
Seems 0243 would be the newest official version? 
https://github.com/mbedmicro/DAPLink/tree/0243

Archcady, 2017-07-13:
cc @tung7970

sg-, 2017-07-13:
This board is not part of mainline DAPLink. Please contact Realtek to report an issue and get a fix.

closed 2017-07-13 by sg-

ErkkiMehtala, 2017-07-14:
You can use mbed-flasher to test this. Currently it prints everything is ok but device is not really reset. For example if using different  blinky binaries you can see if it works. 

C:\workspace>mbedls
+-------------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name     | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+-------------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| REALTEK_RTL8195AM | REALTEK_RTL8195AM[0] | D:          | COM13       | 46000000061488e700000000000000000000000097969902 | 0241            |
+-------------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
C:\workspace>mbedflash -vvv flash -i mbed-os-example-client.bin --tid 46000000061488e700000000000000000000000097969902
[DEBUG](mbed-flasher): [{'target_id_mbed_htm': '46000000061488e700000000000000000000000097969902', 'mount_point': 'D:',
'target_id': '46000000061488e700000000000000000000000097969902', 'serial_port': u'COM13', 
'target_id_usb_id': '46000000061488e700000000000000000000000097969902', 'platform_name': 'REALTEK_RTL8195AM'}]
Going to flash following devices:
46000000061488e700000000000000000000000097969902
[DEBUG](mbed-flasher): [{'target_id_mbed_htm': '46000000061488e700000000000000000000000097969902', 'mount_point': 'D:',
'target_id': '46000000061488e700000000000000000000000097969902', 'serial_port': u'COM13',
'target_id_usb_id': '46000000061488e700000000000000000000000097969902', 'platform_name': 'REALTEK_RTL8195AM'}]
[DEBUG](mbed-flasher): Flashing: 46000000061488e700000000000000000000000097969902
[INFO](mbed-flasher): sendBreak to device to reboot
[INFO](mbed-flasher): reset completed
[DEBUG](mbed-flasher): SHA1: b3a2a6f9a1ca4f6a2a77d9fc2ebf10a19e624e7c
[DEBUG](mbed-flasher): copying file: "C:\workspace\mbed-os-example-client.bin" to D:\mbed-os-example-client.bin
[DEBUG](mbed-flasher): COMMAND: copy "C:\workspace\mbed-os-example-client.bin" D:\mbed-os-example-client.bin
        1 file(s) copied.
[DEBUG](mbed-flasher): copy finished
[INFO](mbed-flasher): sendBreak to device to reboot
[INFO](mbed-flasher): reset completed
[DEBUG](mbed-flasher): verifying flash
[DEBUG](mbed-flasher): ready
[INFO](mbed-flasher): flash ready
[DEBUG](mbed-flasher): dev#1 -> SUCCESS

C:\workspace>mbedflash -vvv reset --tid 46000000061488e700000000000000000000000097969902
[INFO](mbed-flasher): Starting reset for target_id 46000000061488e700000000000000000000000097969902
[INFO](mbed-flasher): Method for reset: simple
[INFO](mbed-flasher): sendBreak to device to reboot
[INFO](mbed-flasher): reset completed