ARMmbed/mbed-os #4757: efm32pg_stk3401 Export broken

chrissnow, 2017-07-13:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
efm32pg_stk3401

**mbed-cli version:** 
1.2

**Expected behavior**
Export should work

**Actual behavior**
Nothing exports
**Steps to reproduce**
```
mbed export -m efm32pg_stk3401 -i simplicityv3
Scan: .

mbed export -m efm32pg_stk3401 -i uvision5
Scan: .
```
----------------------------------------------------------------

theotherjimmy, 2017-07-13:
@chrissnow Your steps to reproduce look like they include correct behavior. Could you provide more detail on what you expected to see that is missing?

chrissnow, 2017-07-17:
@theotherjimmy it didn't export the project, but now I cant reproduce it so it may be something completely unrelated...

I will close until I can reproduce again.

closed 2017-07-17 by chrissnow